AxisCommunications/media-stream-library-js #36: Stream interruption detection

Smorka, 2018-05-25:
I'm using the WebSocketRtspVideo pipeline to stream video. 
However, if there is a connection interruption it seems the player just freezes.
Is there a way or event to detect that the stream/socket connection breaks? So that I may take action or inform the user?

steabert, 2018-06-01:
The pipelines should expose errors that happen on the component. A solution would be to have an .onerror handler on the pipeline that is called when an error happens in a component.

steabert, 2018-06-01:
Note that we can't detect all connection problems, so this won't necessarily fix your particular problem. Do you have any more specific information about the type of connection problem you want to handle?

yellowiscool, 2018-10-05:
We would like to be sure to have an always playing stream, whenever possible.

We do listen on errors from this axis library, or events from the HTML5 media element, notably the `stalled` event. In addition to that, we compare the binary data of the last displayed frame every few seconds to make sure that something is at least moving (we did have stuck video players due to the web browser too).

It would be nice if all this kind of controls / checks could be handled by this library.

steabert, 2019-03-10:
I'm thinking about adding a `onStale` callback on the mse component / html5 video pipeline that would be called when the video is not paused, but playback is not advancing. Still thinking about a good way to go about this.

steabert, 2021-04-10:
Closed since too old. Reopen if someone wants to work on this.

closed 2021-04-10 by steabert