AxisCommunications/media-stream-library-js #42: No MediaSource support with Samsung Internet

yellowiscool, 2018-06-29:
Hei,

We have one user with Samsung Internet  7.2. This browser doesn't support MediaSource by default, so the media-stream-library-js triggers the exception `window.MediaSource is not a constructor`. A more explicit error message such as "this web browser doesn't support media sources" could be a minor improvement.

steabert, 2018-08-03:
Just wanted to mention that there is a function to detect supported formats, which can be used to avoid problems like this.

But sure, a better error message couldn't hurt.

reijosirila, 2019-04-03:
> Just wanted to mention that there is a function to detect supported formats, which can be used to avoid problems like this.
> 

Oh. Where can I find this function ?

steabert, 2021-04-10:
https://developer.mozilla.org/en-US/docs/Web/API/MediaSource/isTypeSupported

closed 2021-04-10 by steabert