AxisCommunications/media-stream-library-js #67: Feature request: Include dist folder when publishing package

gaetancollaud, 2018-07-23:
Is it possible to include the dist folder when publishing ? Your company did this with [locomote](https://github.com/AxisCommunications/locomote-video-player).

Currently when we do a `yarn add` we only get the sources. So when we want to use this lib we have to compile it manually. I'm using angular and I need to include the `min.js` file in my scripts. This way it will be loaded with the app.

Adding dist is easy, we just need a `.npmignore` file at the root. The syntax is the same as for `.gitignore` but it is used only when publishing the lib. [NpmIgnore documentation](https://docs.npmjs.com/misc/developers). This way we can avoid having the dist in github, we only have it in npmjs repository.

If you accept this request, I would gladly make a pull request.

steabert, 2018-08-03:
I don't see any problem with this, pull request welcome.

One of the reasons it's not included is that the `.min.js` file is rather large, since it supports IE, so I considered it to be more of a last resort for people that couldn't build their own bundle.

altiux-admin, 2018-08-13:
media-stream-library.js file is missing in the repository. Could you help me in getting it done.

lekoaf, 2020-11-13:
This issue seems to be resolved.

closed 2020-11-13 by lekoaf