AxisCommunications/media-stream-library-js #74: Stream delay due to buffering

wsain001, 2018-08-07:
I am also using the WebSocketRtspVideo pipeline to stream video for LIVE and real-time video applications.

The problem:
When the WebSocket.readyState != 1 (OPEN), the (read-only) WebSocket.bufferedAmount grows as the message queue grows. This is normal and fine except for LIVE and real-time video applications where the client wants to see what is happening at that moment and it is not interested in lost data. Much like the MJPEG stream works.

Question: 
How do I either flush the queue, or set the queue size to nearly zero. Meaning, I want to tell the Axis camera server not to queue anything. If it cannot send it, then discard the data. In our application, not only are we not interested in the past, but it is imperative that we see what is happening in real time.

ref: https://developer.mozilla.org/en-US/docs/Web/API/WebSocket

steabert, 2018-08-10:
Are you talking about `WebSocket.bufferedAmount` on the client side that is increasing? That would mean that somehow the data doesn't reach the server, maybe a network problem? Can you be more specific about your setup? E.g. URI, resolution, encoding, browser details, latency you observe, etc.

The current latency for H.264 will depend on the browser, with Chrome typically having lowest latency, followed by Firefox. This should be below 500 ms with a good connection.

wsain001, 2018-08-15:
I'm only talking about buffering on the server side. We do not send any web socket data to the axis camera server.
 
Our setup:
Browser: **Chrome**
Encoding: **H264**
Resolution: **720p**
URI: **ws://camera-ip-address/rtsp-over-websocket**
Code: **simple-player.js** from your [this repo](https://github.com/AxisCommunications/media-stream-library-js/blob/master/examples/browser/camera/simple-player.js)
Firmware: 8.x.x
Camera: AXIS M1065-LW
Connection: Wireless
Mobility: Our system is fully mobile and the distance between the camera and the router varies constantly.  

Problem description:
As the camera moves further away form the router the video starts braking up. This is understandable as we reach limits of the router's range. Now, as the camera comes closer, the stream becomes steady again. The problem is that we now see footage that is not longer live and real time, but rather video with seconds of delay. If the browser is refreshed and the web socket connection is re-established, then the delay goes away.

What I think is happening:
When the camera is out of reach and the server cannot send video data to the camera it gets buffered, and as soon as the server is able to send that data it sends the buffered data. This is also understandable in order to prevent data loss. 

What we need:
Much like the MJPEG stream in the camera where JPEGs that are lost are simply lost, we want a similar option for the WebSocket stream. Namely, we want real time footage at ALL times, which would require a mechanisms though which to flush the queue on demand or by configuration.

steabert, 2018-09-14:
Thank you for the detailed explanation.

With respect to the behavior of the web-socket server: since this is a merely a client library, we cannot change anything about the server behavior (that is also why I asked if you were talking about client side buffering).

Your assumption that the web-scoket server is buffering data seems unlikely to me (although I can't say for sure if this could happen). I personally think that when the connection deteriorates, the data comes in slower than the rate at which the video is playing. This might cause the browser to decide to buffer more video (note that you can't really force the browser into a low latency).

Your best option would be to either:
- detect that the connection has re-established and then refresh the stream
- detect that the buffer has increased above a certain threshold, and then increase playback rate until it dropped below the threshold again

It would be best if you keep a debug logging open while testing, so you can see the frames arriving in real time.

yellowiscool, 2018-09-26:
Hei,

I also had latency issues until I realised that using Zip FPS mode, the camera doesn't send I-Frame very often (or not at all if nothing moves) and the player needs one to start playing the video.

wsain001, 2018-11-28:
About your options list:

> detect that the buffer has increased above a certain threshold, and then increase playback rate until it dropped below the threshold again

I can do this client side. However, when the buffer is server side (because there are two buffers - one client side, and one server side) there should be a mechanism through which to tell the server not to buffer, and to simply discard the frames that are not sent. This will keep the feed on the client side real-time.

closed 2020-03-27 by steabert