AxisCommunications/media-stream-library-js #77: Speak through the camera using the computer microphone

gaetancollaud, 2018-08-17:
I'm currently working on replacing flash completely. My company currently use locomote player. This player was able to use the computer's microphone to talk through the camera.

I came up with a POC of this feature using only HTML5 (and a websocket proxy because you cannot open a TCP stream from the browser).

Would it be conceivable to integrate this feature into this repository ? Or do you think it's a completely separate feature ? There is a bit of re-sampling and transcoding to do and I wanted to do this in a service worker so I know that it's not that easy to integrate in this project.

I know that the primary focus of this repos is to stream RTP, but I was just wandering.

steabert, 2018-09-18:
Sure, we welcome any such additions. Even though it says "primary purpose is RTP" in the `README.md` file, there is no reason why the library can't be more than that. As long as you have a feature that fits in the component / pipeline streaming concept, we are happy to include it.

lekoaf, 2020-11-13:
Closing due to inactivity. Feel free to open again if this was an error.

closed 2020-11-13 by lekoaf