AxisCommunications/media-stream-library-js #81: Video element argument missing

farukhzahoor, 2018-09-07:
Hello,

When I am trying to use simple.html sample, I am getting this error. Please help

Uncaught (in promise) Error: video element argument missing
    at new e.exports (media-stream-library.min.js:1)
    at new e.exports (media-stream-library.min.js:1)
    at play (simple-player.js:42)
    at HTMLButtonElement.playButton.addEventListener (simple-player.js:69)

farukhzahoor, 2018-09-14:
Please reply to the above question. I have already talked with Axis support on partner pages. They said we should get answers with GitHub.

steabert, 2018-09-14:
Could you provide:
- specific version you are using
-

steabert, 2018-09-14:
Could you provide:
- specific version you used
- exact steps to recreate the issue

farukhzahoor, 2018-09-14:
- I used v4.0.7
- Check this url https://jmp.sh/QwxTjTN for the zip file with example where we get error. Use simple.html file

steabert, 2018-09-14:
As far as I can see, your `simple.html` and `simple-player.js` files are from version 5 of the library, while your `media-stream-library.min.js` file is from version 4. You cannot mix major version like that, since the API is different. Instead, you need to use examples and library bundle from the same version.

If you somehow found these examples and library bundle together in some place, let me know so I can fix it.

closed 2018-09-18 by steabert