AxisCommunications/media-stream-library-js #88: Delay on Firefox

sanjeyac, 2018-09-19:
Hi,

I noticed a strange delay on Firefox, 
I compared the on-board video client with the simple-player sample, both on Firefox. 

Simple-player sample got stuck for 10 seconds and after that 10 seconds it start to work perfectly, 
but this behavior is not happening on the on-board client.
On the other side this is not happening on Chrome, where the sample works perfectly as expected.

I'm working on a wired local network with an Axis AXIS P1448 (Firmware 8.30.1, Web 2.5.1.15)
I'm using the [simple-player sample](https://github.com/AxisCommunications/media-stream-library-js/blob/master/examples/browser/camera/simple-player.js), in the examples folder.

thanks for any feedback

steabert, 2018-09-19:
Yes, currently there is nothing in the example that makes sure the browser knows it's a live stream. You need to add a `onSourceOpen` callback, which will be called when the MediaSource element has been prepared, like so:

```
pipeline.onSourceOpen = (mse, tracks) => {
  mse.duration = Infinity
}
```
You would add the above to the example, right after you created the pipeline. Does that work for you?
We could add that to the simple player example.

sanjeyac, 2018-09-19:
Hi,

Unfortunately it didn't work,
`mse.duration = Infinity`

but I found this on the [VAPIX Streaming API](https://www.axis.com/files/manuals/vapix_video_streaming_52937_en_1307.pdf)

> duration=<int> /  An unsigned integer  / 0=unlimited

so tried this:

`mse.duration = 0;`

And now it's working perfectly!

Thanks for your help!

Are there any in-depth documentation about this library?

steabert, 2018-09-19:
I ran a check now with Firefox and I actually didn't need to set the `mse.duration` at all, it worked. Since you refer to VAPIX duration, are you playing a continuous recording or something?

sanjeyac, 2018-09-19:
No, all configurations have default values.

steabert, 2018-09-19:
Which version of media-stream-library did you use?

sanjeyac, 2018-09-19:
I compiled the last committed version on master branch

steabert, 2018-09-19:
Can you also add: operating system & version, the version of Firefox, and addition to the example that made it work. Thanks.

sanjeyac, 2018-09-19:
OS: Ubuntu 18.04.1 LTS  - (64bit)
Firefox Quantum: 62.0 (64bit)
Lib commit: 3b022e5

---

I edited the [simple-player.js](https://github.com/AxisCommunications/media-stream-library-js/blob/master/examples/browser/camera/simple-player.js) file at line 46 as follows:

`

    // Setup a new pipeline
    const pipeline = new Pipeline({
        ws: {uri: `ws://${host}/rtsp-over-websocket`},
        rtsp: {uri: `rtsp://${host}/axis-media/media.amp?videocodec=${encoding}`},
        mediaElement
    })

    // notify live stream -- FIX for Firefox
    pipeline.onSourceOpen = (mse, tracks) => {
        console.log("setup mse duration")
        mse.duration = 0
    }

    // play rtsp
    pipeline.ready.then(() => {
        pipeline.rtsp.play()
    })
`

lekoaf, 2020-11-13:
Closing due to inactivity. Feel free to open again if this was an error.

closed 2020-11-13 by lekoaf