AxisCommunications/media-stream-library-js #90: The NPM package build doesn't support Internet Explorer 11

yellowiscool, 2018-09-20:
Hei,

When I use the NPM package build, Internet Explorer cannot load the library.

```javascript
import { pipelines } from 'media-stream-library/lib/index.browser';
```

However, if I download the .min.js file from the Github Releases and import it manually, then it works with Internet Explorer.

Could it be possible to include a IE 11 compatible build inside the NPM package ?

Thanks.

steabert, 2018-09-21:
If you directly include from `lib` in your code, then you need to make sure your own build system will properly add support for different browsers.
However, the NPM package should contain a `dist/media-stream-library.min.js` file which you can import to avoid having to do that yourself, but it seems to be missing. I checked that when packaging locally the file gets included, but somehow it doesn't get included through Travic CI. I'll have to investigate.

steabert, 2018-09-21:
Turns out it wasn't a bug, we hadn't built a new NPM package yet after the `.npmignore` was added. I have pushed a new beta version, so now the latest beta should include the minified file.

yellowiscool, 2018-09-21:
I tried to use @babel/polyfill and @babel/preset-env without success. Did
you do anything special to build the library?

Otherwise yes including the compiled file in the dist folder would work for
me. Is it also possible to include the not minified build?

Thanks

steabert, 2018-09-21:
If you look at our `webpack.config.js`, you'll see a line for babel configuration:
```
presets: [['@babel/preset-env', { 'useBuiltIns': 'usage' }]],
```
As far as I can remember, it's the `useBuiltIns` that is bringing in the polyfills for IE.

The minified file should now be in version `5.0.0-beta.3` of the NPM package.

closed 2019-03-10 by steabert