AxisCommunications/media-stream-library-js #96: The stream delay may grow a lot and will not go back to normal

yellowiscool, 2018-10-05:
Hi,

In some situations (networks latency / issues / dynamic FPS bug), the stream delay may grow a lot, sometimes up to 20 seconds. It never goes back to a normal. The solution is then to restart the stream, which involves some back and forth RTSP commands (it's slow), or to play with the `currentTime` and `buffered` properties of the media element but it's not working all the time and we feel like this should be handled by your library.

Could this library handle high latency / broken stream / stalled streams  ?

steabert, 2018-10-08:
Since this library is not a full media player (as mentioned in the `README.md` file), we don't want to force a certain behaviour on the HTMLVideoElement. For example, the only thing the mse component takes care of is to stream video data to that element. Once that is done, it's up to the browser to display the video, and it's up to the user of the library to control the video element.

Of course, if there is anything wrong with the data stream itself the components should throw an error, or emit an event. If you think a component should be able to handle such a case, please let us know with a specific case/component.

The issues you describe could be monitored by inspecting the video element. If the `currentTime` freezes or if the difference between `currentTime` and `buffered.end(0)` properties on the video element grows too large, you can simply restart the stream (I don't think there is more reliable solution). As you mention yourself, playing with the `currentTime` property in a live stream will not work all the time.
Dynamic FPS will cause latency issues, I haven't seen any browser that can handle that kind of case (you refer to a dynamic FPS bug?).

yellowiscool, 2018-10-08:
Fair enough.

The stream delay issue is more apparent when we tried the Dynamic FPS. So you confirm we shouldn't enable the Dynamic FPS option because it's not well supported by web browsers ?

steabert, 2018-10-08:
Once video data is sent to the browser, it's the browser that is managing playback. The browser will buffer an amount that it thinks is appropriate (hence, there will be differences between browsers). When using dynamic FPS, it will happen that the browser isn't fed any new frames for a while and then it will stall. Once a new frame finally arrives, it will have a long duration, and the browser will continue to display with that duration, causing an increase in latency at best, or a crash at worst.

If you use an Axis camera and want to use dynamic FPS, I recommend you to also use the VAPIX parameter `videoframeskipmode=empty`, in which case the camera will send a "no-difference" (empty) frame instead of no frame at all. This will make sure you receive a constant fps in the client, but with the data savings of dynamic FPS (the empty frame overhead should be negligible for most purposes).

yellowiscool, 2018-10-10:
Thanks for the reply, that's very interesting. I'm planning to collect data about these buffering differences between browsers.

Unfortunately we don't have access to the VAPIX parameters on our AXIS cameras (we use the hosted platform which doesn't support it).

lekoaf, 2020-11-13:
I think it's safe to say that `videoframeskipmode` should be available from LTS 8.40, but probably a bit earlier.

Closing due to inactivity. Feel free to open again if this was an error.

closed 2020-11-13 by lekoaf