AxisCommunications/media-stream-library-js #99: RTSP authentication

jvanvorst, 2018-11-20:
Hi,
Any plans to add support for RTSP authentication?
Thanks.

skeary, 2018-12-02:
@jvanvorst (I think) I hit the same problem you did. I made a change to html5-video-pipeline.js to support authentication and it works for me so may be suitable for you. I've created a PR (#102) but, in the meantime, you could create your own version of html5-video-pipeline.js based on the PR, and use that instead of the current implementation in the library. Note that since I was using a non-Axis camera I also had to make a small change to auth-component.js so this may also be applicable to you? This change is also in the PR.

steabert, 2019-01-02:
Could you please provide details about the exact version you used (git commit SHA) and what error you get?

cherici-simone, 2019-01-03:
Hello, I clone the latest source version (master) and the error is:
_SCRIPT12008: SCRIPT12008: WebSocket Error: Incorrect HTTP response. Status code 401, Unauthorized_
I try to get stream from an AXIS M3027-PVE ip cam. Another thing I do not understand is tokenUri url: if I do not pass any argument the script try to get http://{HOST}/rtspwsswssion.cgi; I had a look (ftp) in the camera cgi files but i did not find any cgi with that name...

cherici-simone, 2019-01-03:
Hello, you are right. The problem is within ie (with chrome work like a sharm). 
I take the example in the repository and just modify the pipeline to include auth configuration.

Gist: https://gist.github.com/cherici-simone/9dcd19eaa599aa02fd86db284edccee9

As you can see I have also change ws configuration to use USER:PWD@ otherwise I get authenticazion error also on chrome. wsHost and host are quite the same: they differ only for the port, because in my environmnet standard port are closed and I have to use 8080 for http/ws and 8081 for https/wss

nikoleh, 2019-01-04:
Hello

I'm facing the same issue with @cherici-simone with both Chrome and Firefox.

Even though AuthComponent takes care of the needed auth headers with RTSP stream (as this works with CliMp4Pipeline), WebSocket request still is unauthorized and in that case AuthComponent becomes irrelevant. 

This is what happens:

> WebSocket connection to 'ws://192.168.1.14/rtsp-over-websocket' failed: HTTP Authentication failed; no valid credentials available

Then the onError handler of WebSocket gets executed, where SOP becomes an issue when using different port or host. When running browser with SOP disabled, the issue is that /axis-cgi/rtspwssession.cgi is not found. Is this endpoint only present in certain firmwares? /rtsp-over-websocket endpoint is there though.

steabert, 2019-03-10:
@nikoleh You shouldn't need the auth component if you're using an Axis camera, but you do need to handle digest authentication. In the examples for a camera, you can find that in `simple-player.js` we're performing an initial auth against `/axius-cgi/usergroup.cgi`, which takes care of the this.
Let me now if it worked or if you need further help.

closed 2019-03-10 by steabert