AxisCommunications/media-stream-library-js #113: Video stream hangs on load

feliciasv, 2019-02-04:
I tried to a html5 Video Pipeline into a React component, based on the example player. But sometimes when the page loads, the video stream seems to hang, only showing the first frame. It never seems to hangs when the the component is remounted though.

This is using media-stream-library: 4.0.7

The (simplified) code causing the problem looks like this:

import { pipelines } from "media-stream-library";
import React, { Component } from "react";

export default class Live extends Component {

    public render() {
        return (
            <video autoPlay={true} />
        );
    }

    public async componentDidMount() {
        this.play(this.props.host, this.props.encoding);
    }

    private play(host, encoding) {
        const videoEl = document.querySelector("video");

        const pipeline = new pipelines.Html5VideoPipeline({
            rtsp: { uri: `rtsp://${host}/axis-media/media.amp?videocodec=${encoding}` },
            videoEl,
            ws: { uri: `ws://${host}/rtsp-over-websocket` },
        });

        pipeline.ready.then(() => {
            pipeline.play();
        });
    }

}

steabert, 2019-02-06:
I wonder if this is a problem with Chrome not allowing the video because there is audio?
You can try to set the muted attribute to false so that the video can play:

```
...
    public render() {
        return (
            <video autoPlay={true} muted={true} />
        );
    }
...
}
```

Be sure to check the `video` element in the console and see if it is paused or not when the frame is shown.

feliciasv, 2019-02-08:
There shouldn't be any audio in the source. But I tried your modification, and I haven't been able to replicate the error after that. So that seems like a fix. Although I still don't understand why this would be needed.

marlun, 2019-02-08:
Most browsers now have policies on when you're allowed to autoplay video:

https://developers.google.com/web/updates/2017/09/autoplay-policy-changes
https://webkit.org/blog/7734/auto-play-policy-changes-for-macos/
https://hacks.mozilla.org/2019/02/firefox-66-to-block-automatically-playing-audible-video-and-audio/

feliciasv, 2019-02-11:
Oh, yes that makes sense.