AxisCommunications/media-stream-library-js #121: audio track in sdp causes crashes

fabianeichfeldt, 2019-02-20:
I've an Foscam camera which provides two media types in SDP:
```
media: 
   [ { type: 'video',
       port: '0',
       proto: 'RTP/AVP',
       fmt: '96',
       connectionData: [Object],
       bwtype: 'AS',
       bandwidth: '96',
       rtpmap: [Object],
       range: 'npt=0-',
       fmtp: [Object],
       control: 'track1' },
     { type: 'audio',
       port: '0',
       proto: 'RTP/AVP',
       fmt: '0',
       connectionData: [Object],
       bwtype: 'AS',
       bandwidth: '64',
       control: 'track2' } ]
```

this causes trouble in
- boxbuilder.js
- mse-component.js
- rtspsession-component.js

by requesting media.rtpmap

**Proposal**: check type of media and skip if it isn't 'video'
I am preparing a PR for this

steabert, 2019-02-22:
Skipping if it isn't 'video' wouldn't work because there are 'audio' encodings we do handle, and even metadata types that can be used. However, I saw in your PR that you just skip if the media does not have an `rtpmap`, which is fine as a work-around.

Better would be to actually handle static payload types, and let the eventual decision happen when the decoder is looking at the media types.
In your case this would mean that we parse the audio type as having `payloadType=0` (from `audio 0 RTP/AVP 0`) and derive the clock rate from a table.

fabianeichfeldt, 2019-07-18:
not reproduceable anymore

closed 2019-07-18 by fabianeichfeldt