AxisCommunications/media-stream-library-js #131: Deconding problem for some camera configurations

epirovano, 2019-03-14:
I'm using the library for testing different camera models, everyone streaming video in H264 format.
All the camera working correctly except one and l've find out that the video is working only if I change the "Entropy coding" from CAVLC to CABAC into camera configuration. In the other case I see a frame, than black image until the next keyframe.

Have you experience similar problem? I'm not an expert in data compression schemas so I don't know wht's the difference between the two configurations.

Thank you

steabert, 2019-03-15:
I can't think of any reason why changing from CAVLC to CABAC would cause a problem, unless the change in setting does more than just changing this.

You can try to log the difference between the streams with `ffprobe` (https://ffmpeg.org/):
```
ffprobe -rtsp_transport tcp -show_frames rtsp://...
```
This will log the frames for each of the streams, you can stop it after a few seconds.
Maybe that will give some more information, you can then attach the files here to a comment.

epirovano, 2019-03-18:
I have compared the two streams with ffprob, I attach there the two output. If you compare the two files you can see that the only difference between the streams are the pkt_pos and pkt_size in each frame.
Another important detail is that if I save on a file the stream right before to push it into the mse, the video play correctly with computer players. The problem is only in the browser.

Thanks 

[output-NOT-working-stream.txt](https://github.com/AxisCommunications/media-stream-library-js/files/2977422/output-NOT-working-stream.txt)
[output-working-stream.txt](https://github.com/AxisCommunications/media-stream-library-js/files/2977423/output-working-stream.txt)

steabert, 2019-03-18:
If the mp4 file plays correctly, it might be the browser. Can you play the saved mp4 file if you drag and drop it into Chrome?

If possible, can you attach both saved mp4 files (working and not-working)?

epirovano, 2019-03-20:
If I drag and drop the two files into chrome they both works...
I attach there the two MP4 files

[streams.zip](https://github.com/AxisCommunications/media-stream-library-js/files/2987164/streams.zip)

Thank you

steabert, 2019-03-28:
The only thing that puzzles me is that the out-of-band decoding data sent through the SPS and PPS are exactly the same for both streams. I would expect some kind of difference in them relating to the CABAC/CAVLC choice, unless that information is somehow present in each frame. I don't know enough about H.264 to know the answer I'm afraid.

Kirkify, 2019-04-02:
I'm actually noticing something very similar, although not related to CABAC/CAVLC.  If I run a certain camera stream on Chrome on my Windows laptop, they work without any problems.  But if I run that same stream on Chrome on my Mac Laptop, it manages to decode and only display the first keyframe then get's into an error state.  @steabert Have you seen cases like this with certain streams, where it works on windows and not on Mac?

steabert, 2019-04-22:
It's certainly possible to see OS-dependent bugs, the particular decoder used will differ between different browsers, different operating systems, and even between different operating system versions. The source of the problem might be very difficult to trace though.

lekoaf, 2020-11-13:
Closing due to inactivity. Feel free to open again if this was an error.

closed 2020-11-13 by lekoaf