AxisCommunications/media-stream-library-js #141: Safari 12.1 ends the stream when it is buffering

fungiboletus, 2019-04-02:
I'm trying to make our player work with Safari. I manage to play the stream if I allow autoplay in the "Settings for this website". However unlike other browsers, when the stream is buffering Safari pauses it and emits a `ended` event.

I'm using media-stream-library-js 5.0.0

fungiboletus, 2019-04-02:
I can ignore the `ended` event when it's safari and with the following code it works (beware it's ugly, it's for testing workarounds):

```javascript
setInterval(() => {
   document.querySelector('video').play();
}, 500);
```

steabert, 2019-04-02:
I'm not sure I understand completely, the video autoplays, but then pauses and an `ended` event is emitted, did I get that part correctly?

Auto-play is tricky, browsers do not completely follow the same rules, but the general behavior now is that you'll be able to play any video if you use the `autoplay` and `muted` attributes on the video tag. Does that make a difference for you?

fungiboletus, 2019-04-02:
Sorry, the autoplay was another issue. It's especially tricky on Safari and
yes muting the video helps but I also have some asynchronous code between
the play click event and the code to start the video when it is ready to
play. I don't think it will ever work with Safari, I need to do the
asynchronous code for a reason. But if you disable the anti autoplay it
works.

I'm not 100% we can ignore the autoplay problems, but if we do the issue is
simple : when the video player is buffering once more after the initial
buffering, safari pauses the video and emits a `ended` event. You can
manually play the video again though, by clicking on the play button of the
default controls, or calling .play()

Other browsers will simply pause the video and play it again automatically
when the buffer is long enough. Without emitting the `ended` event.

Safari also emits the `stalled` events a bit after. I really don't
understand why it emits this `ended` event. It really doesn't make sense
for me.

Also, if you put the video tag in loop mode, it will also loop through the
buffer instead of pausing while buffering. It then doesn't emits the
`ended` event but it's a lot worse 😅

lekoaf, 2020-11-13:
Closing due to inactivity. Feel free to open again if this was an error.

closed 2020-11-13 by lekoaf