AxisCommunications/media-stream-library-js #142: Unable to start a stream on Safari iOS

fungiboletus, 2019-04-02:
When we try to play a stream on Safari iOS, we get the following error : `TypeError: undefined is not a constructor (evaluating 'new window.MediaSource()')` (line 47 of mse-component.js).

fungiboletus, 2019-04-02:
So [the obvious reason is that MediaSource isn't supported on Safari iOS.](https://developer.mozilla.org/en-US/docs/Web/API/MediaSource#Browser_compatibility)

Is there a workaround with this library ? Or does Axis has another library for iOS ?

steabert, 2019-04-02:
Indeed, you are correct, iOS doesn't support MediaSource. There is no work-around in this library other than to use e.g. Motion JPEG.
You can read about it here: https://www.axis.com/sv-se/support/technical-notes/browser-support

fungiboletus, 2019-04-02:
Motion jpeg is only a stream of jpeg pictures right?

Do you know whether it is supported by the axis hosted platform? If so, I
just need to use the motion jpeg pipeline in this library?

When you access to the camera admin web interface on iOS, it can play a
video. Is it motion jpeg there? I tried to inspect the page but safari
crashes when I open the remote developer console...

steabert, 2019-04-04:
> Motion jpeg is only a stream of jpeg pictures right?

Yes, so it's a _lot_ of data depending of course on resolution / frame rate.

> Do you know whether it is supported by the axis hosted platform? If so, I just need to use the motion jpeg pipeline in this library?

As far as I know it should be supported on all Axis cameras, but I don't know about the hosted platform. You should check with Axis/your service provider.

> When you access to the camera admin web interface on iOS, it can play a video. Is it motion jpeg  there? I tried to inspect the page but safari crashes when I open the remote developer console...

It should be, otherwise I wouldn't know how it can play live video :)
There should be a "stream statistics" or "client stream information" in your live video interface which can tell you e.g. the bitrate (which you'll see differs a lot between H.264/MJPEG), as well as the ability to switch between supported formats. Can you access that on iOS?

fungiboletus, 2019-04-04:
I accessed the stream information and indeed it's MJPEG on the iOS.

For the time being, we will not support iOS because MJPEG will use too much data. I guess this issue is not relevant anymore.

Thanks for your comments.

closed 2019-04-04 by fungiboletus