AxisCommunications/media-stream-library-js #145: MJPEG on Safari OS X not working

reijosirila, 2019-04-04:
https://github.com/AxisCommunications/media-stream-library-js/blob/1dcdad4de637a31291a4f7c6ab6de656a29a5475/lib/components/canvas/index.ts#L119

```
window.createImageBitmap is not a function. (In 'window.createImageBitmap(blob)', 'window.createImageBitmap' is undefined)
```

Maybe drawImage for Safari and transferFromImageBitmap for Chrome/Firefox? 

Both versions, 5 and 6

steabert, 2019-04-04:
Indeed, we wrongly assume that `window.createImageBitmap` is available because there is a `bitmaprenderer` context. Instead, we should check for it and use the fallback if it's not available.

reijosirila, 2019-04-04:
For some reason fallback didn't work on a Safari also.

steabert, 2019-04-22:
Do you get a more specific error?

reijosirila, 2019-04-25:
I'll try to debug it as soon as possible and give a feedback.

commit bc2df4bd (2019-08-15): Merge pull request #168 from reijosirila/safari_mjpeg

Fixes #145 MJPEG on Safari OS X
closed 2019-08-15 by steabert