AxisCommunications/media-stream-library-js #157: Subscribe to an Axis camera event stream?

dturton, 2019-05-23:
Hi all, is it possible to use this library to subscribe to an axis camera rtsp event stream? I see a pipeline for the browser but it is possible to do within Node.js?

Any example appreciated!

steabert, 2019-06-10:
Certainly, did you look at the node example player? It should be possible to adapt that one using the metadata pipeline instead. If it doesn't work, you can post what you have so far, then it's easier to help if we see what you're trying to do.

dturton, 2019-06-18:
@steabert Thanks for the reply. Could you point me to the node example using metadata pipeline?

steabert, 2019-06-26:
I meant: you can adapt the player example under node to handle metadata. Two steps: you need to use a URI that will send metadata along, and you need to add an `OnvifDepay` component to your pipeline. If you only want metadata, you can create a pipeline that will only take care of metadata.

If you need more help, let me know what you're trying to do exactly, and then I could try to find some time to push a branch with an example.

dturton, 2019-06-26:
I am trying to use the library for a node.js app that will listen for motion events from Axis cameras and then log the event to a db. I am having difficulting understanding how to use this library to only get the XML for events. I am familiar with URL structure such as axis-media/media.amp?video=0&audio=0&event=on but need help on actually listening for the event.

thx

steabert, 2019-07-07:
There are two components that can be used for this: `BasicDepay` and `OnvifDepay`. The latter is probably easiest to use. When creating the component, you pass a callback in the configuration that will run whenever an XML message comes. You then need to handle the XML data in your callback.

dturton, 2019-07-08:
Could you post a quick example of using BasicDepay or OnvifDepay?

steabert, 2019-07-17:
I've included a pipeline and example on the branch `nodejs-metadata`, but I haven't tested it yet. In any case, the pipeline should be fine and shows the overall idea, you set up an Onvif component, and give it a handler, which will receive the XML messages.

closed 2020-03-27 by steabert