AxisCommunications/media-stream-library-js #158: ES5 and IE 11 support

henrikericsson, 2019-06-05:
Any plans on supporting/including an ES5 / IE 11 compatible bundle?

We'd like to be able to do something like this: import { pipelines } from 'media-stream-library/dist/es5/index.browser';

Attempting to import the pipeline directly from the library  and compiling it ourself doesn't seem to work with our current setup as media-stream-library dependencies throw non-strict mode errors.

henrikericsson, 2019-06-25:
Update:

Now using the stable version 6.0.0. Still no support for IE11 any news when and if this will be added? :/

steabert, 2019-06-26:
Hi, we don't support IE11, but we could certainly add an ES5 target with `tsc`, or build a bundle which could work on more browsers. Is that something you have already tested?

henrikericsson, 2019-06-26:
Yes I've tried to include media-stream-library source files in our own webpack/tsc bundling environment by adding an exclusion for it in node_modules but it didn't work all the way, dependencies to the media-stream-library didnt get transpiled to ES5 correctly. We would like to avoid bundling node_modules as well :)

Great! I will take a look and get back to you.

steabert, 2019-06-27:
Yes, `tsc` doesn't bundle dependencies, so you'll have to configure that yourself somehow with babel + webpack if building yourself.
Alternatively, our bundle should actually work on IE11 (I checked our browserlist and it should include support).

henrikericsson, 2019-06-27:
Hmm maybe we aren't importing it correctly then? This is what we're using now to create a Html5VideoPipeline:

**import { pipelines } from 'media-stream-library/dist/cjs/index.browser';**

This results in a SCRIPT1002: Syntax error in Internet explorer 11 starting at line:

--> **class** AACDepay extends component_1.Tube {
    constructor() {
        let AACPayloadType;
        let hasHeader;
        const incoming = messageStreams_1.createTransform(function (msg, encoding, callback) {

Note that we have over 25 other node_module packages that all get compiled down to es5 fine using Babel and Typescript.

steabert, 2019-06-27:
With bundle I mean importing the `media-stream-library.min.js` file (as in the examples, where you then use the `window.mediaStreamLibrary` object).

If you use the `dist/cjs/index.browser` entrypoint as you wrote above, you need to use babel + webpack to convert to ES5 yourself, you can't import that directly for use with IE11. What is your babel/webpack setup? Do you have any other ES6 packages?

Have you tested to import from `dist/es5` after building on the `sv/es5` branch?

henrikericsson, 2019-06-27:
Yes I've tried using the dist/es5 version, that one gives the same errors as when i tried to bundle it myself from the lib files. It seems like the **class** keywords are gone but for whatever reason it still outputs arrow functions:

--> **args[0].replace(/%[a-zA-Z%]/g, match => {**

steabert, 2019-06-27:
I can't find that arrow function is `dist/es5`, so I guess it's coming from some dependency. What webpack config are you using?

henrikericsson, 2019-06-27:
I just tried the minified (pre-bundled) version as well. It also throws syntax errors in IE11.

Relevant part of the webpack config should be: 

`{
          test: /\.tsx?$/,
          exclude: /node_modules/,
          // exclude: /node_modules\/(?!(media-stream-library)\/).*/,
          use: [
            {
              loader: 'babel-loader',
              options: {
                presets: [
                  ['@babel/env', {
                    useBuiltIns: 'usage',
                    corejs: 3,
                    targets: {
                      browsers: ["last 2 versions", "ie >= 11"]
                    }
                  }],
                  "@babel/typescript",
                  "@babel/react"
                ],
                plugins: [
                  ["@babel/proposal-decorators", { "legacy": true }],
                  "@babel/proposal-class-properties",
                  "@babel/transform-regenerator"
                ],
              },
            }
          ]
        }, `

steabert, 2019-06-27:
It seems that the `debug` package doesn't ship transpiled code: https://github.com/visionmedia/debug/issues/701. That means you shouldn't exclude `node_modules` when transpiling.

henrikericsson, 2019-06-27:
I have tried without excluding as well, it still didnt work. But I can dig into it and see if i can get it working that way but as mentioned above as well, the minified version doesn't work either so even when adding it as the browser example (simple) It still throws errors in IE11.

SCRIPT1010: Expected identifier
media-stream-library.min.js (8, 19705)

Followed by SCRIPT5007: Unable to get property 'pipelines' of undefined or null reference
client.js (125333, 1) 

For some reaspn the window.mediaStreamLibrary object is undefined in IE11. Confirmed working in Edge, Firefox and Chrome.

henrikericsson, 2019-06-28:
I agree including node_modules is something you only want to do as a last resort. Do you think there's a good chance of finding a replacement to the debug package?

steabert, 2019-06-28:
It's not a vital package, so it wouldn't cause problems, but it's a nice approach for debugging purposes :)
I've pushed a new change where I instead resolve to the dist version that comes with the debug package, instead of the source file. Could you try if that works? After `yarn:bundle`, I now get a `media-stream-library.min.js` file without any arrow functions. This approach should even work when you build yourself (you then need to add the same resolve alias to your webpack config).

henrikericsson, 2019-06-28:
The media-stream-library.min.js is confirmed working with IE 11! But i noticed that the es5 bundle output was missing in the latest commit. Maybe it will work with that one and importing it ourselves instead of using the min.js file? That would be much cleaner I think :)

steabert, 2019-06-28:
Oh, sorry, I didn't realize you also still wanted the es5 output, I can re-add that, and write something about the fix used with webpack.

henrikericsson, 2019-06-28:
If it works with the same fix for the minified version then absolutely yes, much better if we can import it normaly without declaring it as an external in webpack :)

henrikericsson, 2019-06-28:
Seems like **tsconfig.es5.json** is missing in the latest version on branch **sv/es5**

commit a0a6f021 (2019-06-27): fix: ES5 support

It seems that at least one package (`debug`) doesn't
provide code transpiled to ES5 through node_modules.
That means we need to include node_modules when
building our bundle to get proper ES5 support.

closes: #158
commit de2456a5 (2019-06-28): fix: ES5 support

It seems that at least one package (`debug`) doesn't
provide code transpiled to ES5 through node_modules.
Since it by default imports from src/, a resolve alias
is added to point to the dist/debug.js file instead.

closes: #158
commit 792d2234 (2019-06-28): fix: ES5 support

It seems that at least one package (`debug`) doesn't
provide code transpiled to ES5 through node_modules.
Since it by default imports from src/, a resolve alias
is added to point to the dist/debug.js file instead.

A new ES5 target is also added at `dist/es5`.

closes: #158
commit c6a814c0 (2019-06-28): fix: ES5 support

It seems that at least one package (`debug`) doesn't
provide code transpiled to ES5 through node_modules.
Since it by default imports from src/, a resolve alias
is added to point to the dist/debug.js file instead.

A new ES5 target is also added at `dist/es5`.

closes: #158
closed 2019-06-30 by steabert