AxisCommunications/media-stream-library-js #165: Browser Example not working- Testing with Axis Camera

vivekallgo, 2019-06-19:
Commit ID: commit 16a38115f98bc3afe2bb10794ea12976f52d1712

I am not able to run the browser examples given in this library with Axis Camera. 

Please find the error message from console.
-----------------------------------------------------------------------------------------
media-stream-library.min.js:1 Uncaught SyntaxError: Unexpected token .

simple-player.js:1 Uncaught TypeError: Cannot destructure property `pipelines` of 'undefined' or 'null'.
    at simple-player.js:1
(anonymous) @ simple-player.js:1
------------------------------------------------------------------------------------------

Please find the error snapshot below:

 [image]

From this log it is clear that there is some problem with the Authentication method. Please help me with this or let me know how I need to provide the credentials.
Let me know if you need any other information.

_____________________________________________________________________________________________________
D:\NodeJS\media-stream-library-js>npm run examples

> media-stream-library@6.0.0 examples D:\NodeJS\media-stream-library-js
> webpack --mode development && http-server examples/browser

Hash: d10d465926ec8dd365a2
Version: webpack 4.34.0
Time: 6068ms
Built at: 06/19/2019 9:43:39 AM
                           Asset      Size  Chunks             Chunk Names
dist/media-stream-library.min.js  1.12 MiB    main  [emitted]  main
Entrypoint main = dist/media-stream-library.min.js
[./lib/index.browser.ts] 202 bytes {main} [built]
[./lib/pipelines/html5-canvas-pipeline.ts] 5.75 KiB {main} [built]
[./lib/pipelines/html5-video-metadata-pipeline.ts] 2.92 KiB {main} [built]
[./lib/pipelines/html5-video-pipeline.ts] 5.15 KiB {main} [built]
[./lib/pipelines/index.browser.ts] 303 bytes {main} [built]
[./lib/pipelines/metadata-pipeline.ts] 4.41 KiB {main} [built]
[./lib/pipelines/pipeline.ts] 5.46 KiB {main} [built]
[./lib/pipelines/rtsp-mjpeg-pipeline.ts] 2.67 KiB {main} [built]
[./lib/pipelines/rtsp-mp4-pipeline.ts] 4.31 KiB {main} [built]
[./lib/pipelines/rtsp-pipeline.ts] 3.75 KiB {main} [built]
[./lib/utils/clock.ts] 2.93 KiB {main} [built]
[./lib/utils/config.ts] 1.49 KiB {main} [built]
[./lib/utils/index.browser.ts] 58 bytes {main} [built]
[./lib/utils/protocols/ntp.ts] 472 bytes {main} [built]
[./lib/utils/scheduler.ts] 7.36 KiB {main} [built]
    + 265 hidden modules
Starting up http-server, serving examples/browser
Available on:
  http://10.0.75.1:8081
  http://192.168.1.114:8081
  http://169.254.45.66:8081
  http://127.0.0.1:8081
  http://172.17.163.65:8081
Hit CTRL-C to stop the server
[Wed Jun 19 2019 09:44:57 GMT+0530 (India Standard Time)] "GET /camera/simple.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/74.0.3729.169 Safari/537.36"
[Wed Jun 19 2019 09:44:57 GMT+0530 (India Standard Time)] "GET /media-stream-library.min.js" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/74.0.3729.169 Safari/537.36"
[Wed Jun 19 2019 09:44:57 GMT+0530 (India Standard Time)] "GET /camera/simple-player.js" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/74.0.3729.169 Safari/537.36"
[Wed Jun 19 2019 09:44:57 GMT+0530 (India Standard Time)] "GET /favicon.ico" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/74.0.3729.169 Safari/537.36"
[Wed Jun 19 2019 09:44:57 GMT+0530 (India Standard Time)] "GET /favicon.ico" Error (404): "Not found"
[Wed Jun 19 2019 09:45:47 GMT+0530 (India Standard Time)] "GET /camera/simple-overlay.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/74.0.3729.169 Safari/537.36"
[Wed Jun 19 2019 09:45:47 GMT+0530 (India Standard Time)] "GET /camera/simple-overlay-player.js" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/74.0.3729.169 Safari/537.36"
_____________________________________________________________________________________________________

Please find the HTTP server logs below:
 [image]

Thanks in advance.

jocdu, 2019-07-02:
Hi vivekallgo,
the media-stream-library.js file in the examples/browser folder is a link to the file in the dist folder. This does not work on windows. I think currently you have to override this file after building the lib. I have updated the package.json script to do this here locally.

vivekallgo, 2019-07-02:
Thank you so much @jocdu for help. It is working for me.

closed 2019-07-02 by vivekallgo