AxisCommunications/media-stream-library-js #176: Error while running the browser example

pransen, 2019-06-28:
On the browser console I see this:
Firefox cannot establish a connection to the server at ws://127.0.0.1:8854/

On the terminal, this is what I see:
"GET /axis-cgi/rtspwssession.cgi?1561702741825" Error (404): "Not found"

Please suggest what am I doing wrong.

fabianeichfeldt, 2019-07-04:
Does the node application run well?

steabert, 2019-07-17:
Please provide detailed steps of what you are doing so that we can be of more help, thanks!

eCodee, 2019-08-02:
I cannot get the exmples work, window.mediaStreamLibrary is "undefined":

[image]

All tests pass. Could you provide me with any help?

[image]

nathanptackett, 2019-09-09:
Just gonna add the browser example also doesn't work for me, and I get the same error.

cubecleveland, 2019-11-21:
did u guys have any accomplishments there, figuring this out ?

steabert, 2019-11-21:
I will try to have a look if I can reproduce this.

cubecleveland, 2019-11-21:
Thanks steabert, while you are at it.. some info about webpack dependencies and installing fresh.
FYI, when u install on new node machine the weback comes with tons of dependency errors....
Still trying to go through it

steabert, 2019-11-22:
I'm not sure if this is related to Windows, but the `dist/media-stream-library.min.js` is accessed through a symlink `examples/browser/media-stream-library.min.js`. In linux, this is basically a text file with the path to the real file. Judging from https://stackoverflow.com/questions/5917249/git-symlinks-in-windows, it might be that this path is the content read by the browser (hence the first error message you see). If you open the file `examples/browser/media-stream-library.min.js` do you see a string `../../dist/media-stream-library.min.js`? If so, that will be the problem. You could try to manually copy the `dist/media-stream-library.min.js` file to `examples/browser/media-stream-library.min.js` (overwriting the symlink).

Note that I haven't tried to run `yarn examples` on a Windows machine, maybe someone else that was successful with this can add some info?

If I get some time, I'll try to have a look at this, but for a proper solution to this symlink problem on Windows I'm happy for any helpful contributions.

steabert, 2019-11-22:
@cubecleveland please open a new issue for each problem you might have (in this case what command you ran and the log output). Even if it seems related, it's easier to follow the comments if they cover a specific problem.

PR #543 (2021-01-13): feat: copy minified file to example
Instead of using a symlink, which doesn't work on Windows, we copy the minified file to the examples folder.

Fixes #412
Fixes #391
Fixes #176
commit 2ff3a611 (2021-01-13): feat: copy minified file to example

Instead of using a symlink, which doesn't work on
Windows, we copy the minified file to the examples
folder.

Fixes #412
Fixes #391
Fixes #176
closed 2021-01-13 by steabert