AxisCommunications/media-stream-library-js #182: with Hikvision cameras MJPEG over RTSP not working

reijosirila, 2019-07-11:
Throws 
`
Uncaught Error: no quantization header present
 `
[image]

steabert, 2019-07-17:
I'm guessing there is no in-band quantization data, which is a situation which is not implemented. Could you modify the jpegdepay parser code to log the `type` and `Q` values?

On the other hand, it might be that the quantization tables _are_ send in-band, but not for every frame, in which case the solution should be relatively simple.

In any case, if the quantization data needs to be computed, I welcome any contributions :)

reijosirila, 2019-07-18:
Three different Q values. 
[image]
[image]
[image]

reijosirila, 2019-07-19:
I did quick test like appendix A sais on a link you suggested: 

parser.ts
```
import {
  makeImageHeader,
  makeHuffmanHeader,
  makeScanHeader,
  makeQuantHeader,
  makeFrameHeader,
} from './headers'
import { payload } from '../../utils/protocols/rtp'

/*
 * Table K.1 from JPEG spec.
 */
const jpeg_luma_quantizer = [
  16, 11, 10, 16, 24, 40, 51, 61,
  12, 12, 14, 19, 26, 58, 60, 55,
  14, 13, 16, 24, 40, 57, 69, 56,
  14, 17, 22, 29, 51, 87, 80, 62,
  18, 22, 37, 56, 68, 109, 103, 77,
  24, 35, 55, 64, 81, 104, 113, 92,
  49, 64, 78, 87, 103, 121, 120, 101,
  72, 92, 95, 98, 112, 100, 103, 99
]

/*
 * Table K.2 from JPEG spec.
 */
const jpeg_chroma_quantizer = [
  17, 18, 24, 47, 99, 99, 99, 99,
  18, 21, 26, 66, 99, 99, 99, 99,
  24, 26, 56, 99, 99, 99, 99, 99,
  47, 66, 99, 99, 99, 99, 99, 99,
  99, 99, 99, 99, 99, 99, 99, 99,
  99, 99, 99, 99, 99, 99, 99, 99,
  99, 99, 99, 99, 99, 99, 99, 99,
  99, 99, 99, 99, 99, 99, 99, 99
]

/*
 * Call MakeTables with the Q factor return Buffer
 */
function MakeTables(q: number) {
  let factor = q
  const buf = new Buffer(128)

  if (q < 1) factor = 1
  if (q > 99) factor = 99
  if (q < 50) q = Math.floor(5000 / factor)
  else q = 200 - factor * 2

  for (let i = 0; i < 64; i++) {
    let lq = Math.floor((jpeg_luma_quantizer[i] * q + 50) / 100)
    let cq = Math.floor((jpeg_chroma_quantizer[i] * q + 50) / 100)
    /* Limit the quantizers to 1 <= q <= 255 */
    if (lq < 1) lq = 1
    else if (lq > 255) lq = 255
    if (cq < 1) cq = 1
    else if (cq > 255) cq = 255
    buf.writeUInt8(lq, i)
    buf.writeUInt8(cq, i + 64)
  }
  return buf
}
/**
 * Each packet contains a special JPEG header which immediately follows
 * the RTP header.  The first 8 bytes of this header, called the "main
 * JPEG header", are as follows:*
[... 116 lines cut ...]
    const quantHeader = makeQuantHeader(precision, qTable)

    if (metadata.DRI !== 0) {
      throw new Error('not implemented: DRI')
    }

    const frameHeader = makeFrameHeader(width, height, type)

    return {
      size: { width, height },
      data: Buffer.concat([
        IMAGE_HEADER,
        quantHeader,
        frameHeader,
        HUFFMAN_HEADER,
        SCAN_HEADER,
        ...fragments,
      ]),
    }
  }
}
```

But picture quality is horrible cmpared to camera that includes quantization data. Also VLC streams much smoother picture. 
Maybe somebodys eye can see the issue?

reijosirila, 2019-07-19:
Two pictures compared; same stream. 
[image]

reijosirila, 2019-07-19:
:D Chatting with myself. 
Used tables from here: https://github.com/libav/libav/blob/c4642788e83b0858bca449f9b6e71ddb015dfa5d/libavformat/rtpdec_jpeg.c#L42

And the result is stunning: 
[image]

I'll do Pull Request after code formatted.

reijosirila, 2019-07-19:
Okey, RFC sample code seems to have a mistake. RFC sais "Each table is an array of 64 values given in zig-zag order, identical to the format used in a JFIF DQT marker segment." but sample  does not use zig-zag ordered table from IEC 10918-1 Annex K. So PR uses correct table.

steabert, 2019-07-23:
It seems you get by just chatting with yourself :) Thanks for all the info and the PR, I'll check it ASAP and look at your sources. In the meanwhile, consider including some of your comments in the code.

PR #193 (2019-07-24): feature: added support for Q < 128 to RTP-JPEG
For Q values less than 128, the luma and chroma
quantization tables are computed from the Q value.

Fixes #182.

Created new pull request with new commit message and other requested improvements
closed 2019-07-24 by steabert