AxisCommunications/media-stream-library-js #184: MVHD Box - duration is set to 0xffffffff

fabianeichfeldt, 2019-07-16:
Hi,

I have added a sink which writes the data to disk. This was pretty easy and straight forward. But 
 by replaying this, with VLC I recognized an undefined strange length of the video. (-1243.34, or something like this.)
After some research and experimenting with [mp4Parser](http://mp4parser.com/) and comparing the file with a recording from ffmpeg. I found an difference in the MVHD box.
FFmpeg sets the duration flag to 0. 
We are setting this to 0xffffffff.
After changing it to 0, my replay works as expected also :-)

So my question is, would there be any drawback by changing this duration to 0?
I couldn't found anything about the "undefined" default here: [spec - file format](https://l.web.umkc.edu/lizhu/teaching/2016sp.video-communication/ref/mp4.pdf)

steabert, 2019-07-17:
You're referencing the 2005 version of 14496-12, but if you look at the 2012 version of the spec, it says:

> `duration` is an integer that declares length of the presentation (in the indicated timescale). This property is derived from the presentation’s tracks: the value of this field corresponds to the duration of the longest track in the presentation. If the duration cannot be determined then duration is set to all 1s.

As far as I can recall, not all mp4 players handle fragmented mp4 equally well, so this seems to be something that should be reported to VLC. If you file a bug report with sample mp4 file they should be able to help out.

The reason I would be reluctant to change this to `0` is that even though it might work for VLC, there are other players that might not interpret the value as unkown duration. So it's better to try to report this to the media players first. If it's imperative to have this, you could also transform the duration in your sink component (I use the `.parse` method of the `Box` class to convert back to an object and then modify the data and write it back to a buffer).

fabianeichfeldt, 2019-07-18:
I am not sticking to VLC, it was just my preferred quick test tool. 
So I would close this issue, thanks for your insights.

closed 2019-07-18 by fabianeichfeldt