AxisCommunications/media-stream-library-js #185: unknown chunks in RTSP-Parser

fabianeichfeldt, 2019-07-16:
Hi,

randomly I have an Exception in RTSP-Parser method _setState(), because the first chunk is either an IDLE, INTERLEAVED or RTSP chunk.
It is quite hard to log this exception but I could log 2 chunks which result in the exception:
`1d c4 e9 db ca b4 c9 64 66 e4 4f d6 79 0e 3b 0e da c8 48 cf 75 51 4b 8f 86 c6 12 35 22 75 19 71 41 0a e1 d7 53 0d fa 69 e5 89 55 f7 10 bd ad ee 26 11`  
and 
`ac b7 61 92 29 17 0f 34 2c 70 43 ef 35 b7 cb 38 5d 82 09 e7 3e c4 e1 23 68 3a bd 13 bb 4f a0 3e 0e df 3e 83 ac 66 c5 c2 73 25 cb 9d cd 94 5a 14 c9 76`

Does anybody know what this chunk could be? 
I am running several camera streams in parallel, but I don't know if this is important to reproduce the error. I will try this at night with a single camera.

How would a more resistent implementation looks like? May be just ignoring such bad chunks?

steabert, 2019-07-17:
Interesting, it might have to do with the comment I wrote on line 124 of the parser:
```
// Body is assumed to be the remaining data of the last chunk.
```
which might not be a correct assumption. In that case the error would occur only when you get an RTSP response (when starting to play or when sending RTSP in-between). I marked this as a bug, as we should really be checking the `Content-Length` header to determine body size.

It's difficult to recover from this, as technically there is no way of knowing when the next chunk starts after you get a failure. In that case the error should be handled higher up by e.g. restarting the stream.

RHammar, 2019-09-21:
Fond something similar wich was solved in #240

fabianeichfeldt, 2020-09-15:
> Fond something similar wich was solved in #240

solved with #240

closed 2020-09-15 by fabianeichfeldt