AxisCommunications/media-stream-library-js #211: Importing pipelines from dist/es5 throws syntax errors in IE11

henrikericsson, 2019-08-12:
**Describe the bug**
Using components from dist/es5 outputs es6+ syntax and does not work with IE11

**To Reproduce**
Import and attempt to use components in dist/es5 folder.

Steps to reproduce the behavior:
1. Set up a typescript/webpack project targeting ES5 output

2. Add this to whatever code is going to run on the page (We're using React):
**import { pipelines } from 'media-stream-library/dist/es5/index.browser';**

3. Navigate to the page using IE11 and it should throw a syntax error.

**Screenshots**
[image]

**Environment (please complete the following information):**
 - OS: Windows
 - Browser Internet Explorer 11

**Additional context**

**our tsconfig:**
`{
  "compilerOptions": {
    "esModuleInterop": true,
    "baseUrl": ".",
    "strict": true,
    "preserveConstEnums": true,
    "resolveJsonModule": true,
    "target": "es5",
    "module": "commonjs",
    "outDir": "dist",
    "sourceMap": true,
    "experimentalDecorators": true,
    "suppressImplicitAnyIndexErrors": true,
    "strictNullChecks": false,
    "jsx": "react"
  },
  "files": [
    "./node_modules/@types/node/index.d.ts",
    "./node_modules/@types/jest/index.d.ts",
  ],
  "paths:": {
    "lib/*": ["./lib/*"],
    "src/*": ["./src"],
  },
  "include": [
    "src/**/*"
  ],
  "exclude": [
    "node_modules"
  ]
}`

**relevant parts of our webpack.config:**
`module: {
      rules: [
        {
          test: /\.tsx?$/,
          exclude: /node_modules/,
          use: [
            {
              loader: 'babel-loader',
              options: {
                presets: [
                  ['@babel/env', {
                    useBuiltIns: 'usage',
                    corejs: 3,
                    targets: {
                      browsers: ["last 2 versions", "ie >= 11"]
                    }
                  }],
                  "@babel/typescript",
                  "@babel/react"
                ],
                plugins: [
                  ["@babel/proposal-decorators", { "legacy": true }],
                  "@babel/proposal-class-properties",
                  "@babel/transform-regenerator"
                ],
              },
            }
          ]
        }, ...`

steabert, 2019-08-12:
I wonder if this is the same bug that occurred when we took in actual IE11 support with #175 ? We needed to alias the debug module in the webpack configuration. You can check the `README.md` file for details.

henrikericsson, 2019-08-13:
Missed the part where u had to add the debug as an alias, seems like it's working now. Thanks! :)

closed 2019-08-13 by henrikericsson