AxisCommunications/media-stream-library-js #215: new Html5VideoPipeline(...) throws error in IE11

henrikericsson, 2019-08-14:
**Describe the bug**
Creating a new Html 5 Video Pipeline throws the error:

**SCRIPT5022: insertion point not part of pipeline**

**To Reproduce**
create a new Html5VideoPipeline object imported from :

'media-stream-library/dist/es5/pipelines/html5-video-pipeline';"

And run it in Internet Explorer 11.

**Screenshots**
[image]

**Environment (please complete the following information):**
 - OS: Windows
 - Browser Internet Explorer 11

**Additional context**
Same code works in other browser environments.

steabert, 2019-08-14:
Looking at [Set](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Set), it seems IE11  does not support `new Set(iterable)`, which is what we use in the pipeline constructor to initialize a pipeline.

My guess is you need to include a correct polyfill somehow. Do you get the same error with the minified bundle we ship with the library?

henrikericsson, 2019-08-16:
You are correct IE11 does not support Set, it's part of the ECMA 2015 specifications. We are using babel-polyfill with the current configuration:

`use: [
            {
              loader: 'babel-loader',
              options: {
                presets: [
                  ['@babel/env', {
                    useBuiltIns: 'usage',
                    corejs: 3,
                    targets: {
                      browsers: ["last 2 versions", "ie >= 11"]
                    }
                  }],
                ],
                plugins: [
                  ["@babel/proposal-decorators", { "legacy": true }],
                  "@babel/proposal-class-properties",
                  "@babel/transform-regenerator"
                ],
              },
            }
          ]`

This should cover Set and a multitude of other "new" additions to javascript in ECMA 2015 and above.

The minified worked as of 6.0.0 haven't tried the latest.

steabert, 2019-08-20:
If the provided minified bundle works, I have no idea what the problem could be. Rebuilding the bundle with your browser targets gives essentially the same file, with a `Set` polyfill. Have you examined your resulting bundle?

henrikericsson, 2019-08-26:
Got it to work now, it was a mistake on our end, pollyfilling with babel using the 'usage' option does not include usages in the node_modules folder. We are not using Set anywhere in our own code and as a result babel decides not to polyfill it.

closed 2019-08-26 by henrikericsson