AxisCommunications/media-stream-library-js #231: Mp4Muxer onSync never gets called

PhilDore11, 2019-09-05:
**Describe the bug**
When overriding the `onSync` method of the mp4muxer component, it never gets called even if `ELEMENTARY` messages get transformed

**To Reproduce**

1. Create an `Mp4Muxer` component and supply an `onSync` method:
```
const mp4Muxer = new Mp4Muxer();
mp4Muxer mp4Muxer.onSync = ntpPresentationTime => {
  if (this.onSync) {
    this.onSync(ntpPresentationTime);
  }
};
```
2. Make use of the `Mp4Muxer` component in your pipeline.
3. Notice how, even if the `Mp4Muxer` component transforms messages of type `ELEMENTARY`, the `onSync` method doesn't get called

**Possible solution**
Seems to be an issue with the conditions around the `onSync` method:
https://github.com/AxisCommunications/media-stream-library-js/blob/65943e1b2e87602fc2702c94c2c014175c573df9/lib/components/mp4muxer/index.ts#L57
_ISSUE: The `boxBuilder.ntpPresentationTime` needs to be both `falsy` and `truthy` for the `onSync` method to get called (impossible scenario)_

steabert, 2019-09-06:
The example code you give in `1.` doesn't make sense as how one would use `onSync`. There is also no impossible scenario as far as I can see. Please present an actual code examples that reproduces the issue, while also logging the NTP timestamps from the stream.

PhilDore11, 2019-09-06:
@steabert Thanks for the quick comment. Sorry if I wasn't clear.
Notice this code in `Mp4Muxer`:
```
if (!boxBuilder.ntpPresentationTime) {
    boxBuilder.setPresentationTime(trackId, ntpTimestamp)
    if (boxBuilder.ntpPresentationTime) {
        onSync(boxBuilder.ntpPresentationTime)
    }
}
```
Am I wrong in saying that `onSync` will never get called since `!boxBuilder.ntpPresentationTime` && `boxBuilder.ntpPresentationTime` both need to be `true` for `onSync` to get called?

steabert, 2019-09-06:
Yes, you would be wrong, as the code in between might affect the presence of that property.

If you really don't get the `onSync` to fire, please try to investigate if you can log anything at all at those lines to help with debugging.

PhilDore11, 2019-09-06:
Ok, So is the intension to only call `onSync` on the first `ELEMENTARY` message? Because that is the current behavior (i.e. once `boxBuilder.ntpPresentationTime` is defined, `onSync` doesn't get called anymore)

steabert, 2019-09-09:
Exactly, as soon as the first message with an NTP timestamp arrives, we can compute the `ntpPresentationTime` from this and then report it to the user, to be used e.g. in synchronization calculations. From the `ntpPresentationTime` and video `currentTime` you can always compute the NTP time of the currently displayed image.

PhilDore11, 2019-09-09:
Thanks for all your help @steabert.
Correct me if I'm wrong, but we don't have access to the `ntpPresentationTime` from the video tag right? Not sure how you're suggesting to compute teh NTP of the displayed image.

closed 2019-09-13 by PhilDore11