AxisCommunications/media-stream-library-js #232: Add onSync method to the MseSink component

PhilDore11, 2019-09-05:
**Describe the solution you'd like**
Replicate what the `Mp4Muxer` and `CanvasSink` components do to allow the pipeline to know when it was updated.

steabert, 2019-09-06:
The `onSync` method is tied to `Mp4Muxer` and `CanvasSink`, there is no logical reason to add this to `MseSink`.

closed 2019-09-06 by steabert

PhilDore11, 2019-09-06:
We are looking for a way to hook into when the `MseSink` component updates his state (and get the video time associated)
I notice that the component exposes a `currentTime`, but this doesn't work for us since we are streaming video starting at a specific time. In our case, getting 2.32s from the `MseSink` component isn't useful since we need to use the time from the stream.

steabert, 2019-09-06:
I'm not sure if I understand you.

Anyway, the `onSync` method is meant to be only fired once, as soon as the NTP timestamp of the start of the presentation is known, so that one can e.g. synchronize other stuff with the video.
To get the actual NTP timestamp of the current displayed image, you would then add the `currentTime` to the `ntpPresentationTime`.

I'm pretty sure there is something like that in one of the examples (can't check right now). But you seem to have added a bug report that `onSync` doesn't work for you. I would suggest to investigate that first, make sure your stream is getting RTCP messages and that there is an NTP timestamp on some messages after a while.

jjfrankovich, 2019-11-14:
I am also interested in getting a callback when a new stream image has been painted for a canvas/mjpeg stream.

There is an example which attempts to show how to use the Scheduler and "component.Component.peek" to get a callback when the canvas is updated, but component.Component doesn't seem to be a valid object anymore:
Can you update the example or provide a good mechanism for knowing when the image has been updated?

steabert, 2019-11-15:
The example with the peek component only serves to insert the scheduler somewhere in the stream (there are different ways to do that, e.g. you can run the scheduler as part of a handler for a metadata stream).

But to me it seems that the callback you are after is just the `draw` function which is used to update the image, or? That function is provided by you and will be called when the image should be drawn. The peek component passes each message is it comes over the stream to the `run` method of the scheduler. In turn, the message gets stored in the scheduler. When the message timestamp lines up with the presentation time, the `draw` callback is invoked with the stored message.