AxisCommunications/media-stream-library-js #238: Use index.browser

PhilDore11, 2019-09-13:
We're using this library as part of a react app and have been struggling to use the `browser` target in the `package.json`.
We need to include the `browser` target since we are using `Source` and `Sink`, which are not included in the `main` target (i.e. `index.node.js`)

Our current workaround is to import the `index.browser.js` explicitly:
```import { pipelines, components } from "media-stream-library/dist/esm/index.browser";```

Any help would be appreciated :)

PhilDore11, 2019-09-17:
Could this be related?
https://github.com/webpack/webpack/issues/4674

steabert, 2020-03-30:
Oh, I totally missed this question, sorry!

The `Source` and `Sink` components should be exported with `index.node.js` (that exports everything from `components.ts`). It's only specific components that use the browser's `WebSocket` or `MediaSource` that are not exported.

What kind of error do you get?

lekoaf, 2020-11-13:
@PhilDore11 If you are using this library to stream from AXIS devices, maybe [media-stream-player](https://github.com/AxisCommunications/media-stream-player-js) could be of interest to you. It's a videoplayer written in React with this library in the background.

Anyway, closing this issue due to inactivity. If you feel like that is an error, feel free to open it again.

closed 2020-11-13 by lekoaf