AxisCommunications/media-stream-library-js #260: [Browser] How to import this library as a module ?

fungiboletus, 2019-10-18:
```javascript

import {components, pipelines} from 'media-stream-library';

import {components, pipelines} from 'media-stream-library/media-stream-library/dist/esm/index.browser';

import {components, pipelines} from 'media-stream-library/media-stream-library/dist/cjs/index.browser';

import {components, pipelines} from 'media-stream-library/media-stream-library/dist/es5/index.browser';

import 'media-stream-library/dist/media-stream-library.min'
```

Using vue-cli, webpack, babel and fancy `transpileDependencies` options, I havn't managed to load it.

We have more than 1500 NPM packages as dependencies and only `media-stream-library` can't be loaded.

```html
  <script src="media-stream-library.min.js"></script>
```
However this one does.

icholy, 2020-03-31:
@fungiboletus did you ever figure this out. @steabert can this be re-opened?

steabert, 2020-04-01:
There are two targets in the `package.json` file, and the "default" (`main`) is for NodeJS. However, if you build using e.g. webpack, it should pick up the `browser` target instead, but it seems your build setup doesn't do that?

As a workaround, you can import from `media-stream-library/dist/esm/index.browser.js`, but for a more proper solution, we would need to find out why angular is not picking up the `browser` target.

icholy, 2020-04-01:
I've created this repository which demonstrates the issue: https://github.com/icholy/issue260

I think the original error I encountered was just typescript not respecting the "browser" field.

However, there's another issue. `ng build` gives the following error over and over:

```
ERROR in ./node_modules/hash-base/index.js
Module not found: Error: Can't resolve 'stream' in '/home/icholy/issue260/node_modules/hash-base'
resolve 'stream' in '/home/icholy/issue260/node_modules/hash-base'
  Parsed request is a module
  using description file: /home/icholy/issue260/node_modules/hash-base/package.json (relative path: .)
    Field 'browser' doesn't contain a valid alias configuration
    resolve as module
      looking for modules in /home/icholy/issue260
        using description file: /home/icholy/issue260/package.json (relative path: .)
          Field 'browser' doesn't contain a valid alias configuration
          using description file: /home/icholy/issue260/package.json (relative path: ./stream)
            no extension
              Field 'browser' doesn't contain a valid alias configuration
              /home/icholy/issue260/stream doesn't exist
            .ts
              Field 'browser' doesn't contain a valid alias configuration
              /home/icholy/issue260/stream.ts doesn't exist
            .tsx
              Field 'browser' doesn't contain a valid alias configuration
              /home/icholy/issue260/stream.tsx doesn't exist
            .mjs
              Field 'browser' doesn't contain a valid alias configuration
              /home/icholy/issue260/stream.mjs doesn't exist
            .js
              Field 'browser' doesn't contain a valid alias configuration
              /home/icholy/issue260/stream.js doesn't exist
            as directory
              /home/icholy/issue260/stream doesn't exist
      /home/icholy/issue260/node_modules/hash-base/node_modules doesn't exist or is not a directory
      /home/icholy/issue260/node_modules/node_modules doesn't exist or is not a directory
      /home/node_modules doesn't exist or is not a directory
      /node_modules doesn't exist or is not a directory
      looking for modules in /home/icholy/issue260/node_modules
        using description file: /home/icholy/issue260/package.json (relative path: ./node_modules)
          Field 'browser' doesn't contain a valid alias configuration
      looking for modules in /home/icholy/node_modules
        No description file found
        Field 'browser' doesn't contain a valid alias configuration
          using description file: /home/icholy/issue260/package.json (relative path: ./node_modules/stream)
            no extension
              Field 'browser' doesn't contain a valid alias configuration
              /home/icholy/issue260/node_modules/stream doesn't exist
            .ts
              Field 'browser' doesn't contain a valid alias configuration
              /home/icholy/issue260/node_modules/stream.ts doesn't exist
            .tsx
              Field 'browser' doesn't contain a valid alias configuration
              /home/icholy/issue260/node_modules/stream.tsx doesn't exist
            .mjs
              Field 'browser' doesn't contain a valid alias configuration
              /home/icholy/issue260/node_modules/stream.mjs doesn't exist
            .js
              Field 'browser' doesn't contain a valid alias configuration
        No description file found
        no extension
          Field 'browser' doesn't contain a valid alias configuration
              /home/icholy/issue260/node_modules/stream.js doesn't exist
            as directory
              /home/icholy/issue260/node_modules/stream doesn't exist
          /home/icholy/node_modules/stream doesn't exist
        .ts
          Field 'browser' doesn't contain a valid alias configuration
          /home/icholy/node_modules/stream.ts doesn't exist
        .tsx
          Field 'browser' doesn't contain a valid alias configuration
          /home/icholy/node_modules/stream.tsx doesn't exist
        .mjs
          Field 'browser' doesn't contain a valid alias configuration
          /home/icholy/node_modules/stream.mjs doesn't exist
        .js
          Field 'browser' doesn't contain a valid alias configuration
          /home/icholy/node_modules/stream.js doesn't exist
        as directory
          /home/icholy/node_modules/stream doesn't exist
[/home/icholy/issue260/stream]
[/home/icholy/issue260/stream.ts]
[/home/icholy/issue260/stream.tsx]
[/home/icholy/issue260/stream.mjs]
[/home/icholy/issue260/stream.js]
[/home/icholy/issue260/node_modules/hash-base/node_modules]
[/home/icholy/issue260/node_modules/node_modules]
[/home/node_modules]
[/node_modules]
[/home/icholy/node_modules/package.json]
[/home/icholy/issue260/node_modules/stream]
[/home/icholy/issue260/node_modules/stream.ts]
[/home/icholy/issue260/node_modules/stream.tsx]
[/home/icholy/issue260/node_modules/stream.mjs]
[/home/icholy/node_modules/stream/package.json]
[/home/icholy/issue260/node_modules/stream.js]
[/home/icholy/node_modules/stream]
[/home/icholy/node_modules/stream.ts]
[/home/icholy/node_modules/stream.tsx]
[/home/icholy/node_modules/stream.mjs]
[/home/icholy/node_modules/stream.js]
 @ ./node_modules/hash-base/index.js 3:16-33
 @ ./node_modules/md5.js/index.js
 @ ./node_modules/media-stream-library/dist/esm/components/auth/digest.js
 @ ./node_modules/media-stream-library/dist/esm/components/auth/index.js
 @ ./node_modules/media-stream-library/dist/esm/pipelines/html5-video-pipeline.js
 @ ./node_modules/media-stream-library/dist/esm/pipelines/index.browser.js
 @ ./node_modules/media-stream-library/dist/esm/index.browser.js
 @ ./src/app/app.component.ts
 @ ./src/app/app.module.ts
 @ ./src/main.ts
 @ multi ./src/main.ts
```

icholy, 2020-04-01:
I setup a barebones webpack project and that worked. :/

steabert, 2020-04-01:
The `stream` module is only present in Node but will get replaced with e.g. `stream-browserify` by webpack if you bundle for the browser (it does that automatically).

You might have more success importing from `media-stream-library/dist/media-stream-library.min.js`, which is already bundled by webpack.
If that doesn't work, you'll have to figure out how to polyfill the Node APIs with Angular.

fungiboletus, 2020-04-02:
@icholy I was using `media-stream-library.min.js` for some time. It's not perfect because it contains polyfills twice but it works.

I use this library with nodejs on the server side and I'm very happy about it, but I eventually reimplemented the browser parts because it was more convenient and I could have more control.

icholy, 2020-04-02:
@fungiboletus 

> I was using media-stream-library.min.js

Yeah, I think I'm going to go down this road too.

> It's not perfect because it contains polyfills twice but it works.

What do you mean by this? The bundle/build contains duplicate polyfills?

fungiboletus, 2020-04-03:
Yes media-stream-library.min.js will contain polyfills for itself, and then your app contains polyfills for itself too. So you may have the same polyfill twice, but it's not a big problem I think.

lekoaf, 2020-11-13:
Closing this issue due to inactivity. If you feel like this was an error, please reopen it.

closed 2020-11-13 by lekoaf