AxisCommunications/media-stream-library-js #290: Using The library

cubecleveland, 2019-11-20:
HI Guys,

Sorry for the ingnorance but im trying to get an idea of how to use the library.
We installed via node, but when trying to run examples get build errors.

IT is very unclear from the library what is the step by step process to have this working.. I am assuming there should be a server running in the backgrounf ?
A little help will be very much appreciated

steabert, 2019-11-22:
A first step would be to post the exact steps + output here, otherwise it's difficult to guess what you are trying to do :)

There isn't really a detailed step-by-step, the current documentation assumes that you're familiar with web/node development. However, with your help I'm happy to try and improve the instructions so that you can get going. A first step would be to explain your use case, what you're trying to accomplish, etc.

cubecleveland, 2019-11-22:
Absolutly. and thanks for the response!

i am at a phase where im simply trying to run the library :) but can not compile the webpack.
So simply running run examples crashes....

mkdir media
npm install media-stream-library
cd node_modules
cd media-stream-library
yarn examples
[image]

steabert, 2019-11-22:
Yes, you can't do that, you should only run the development environment from a git repo where you installed dependencies.

To run the examples directly from the installed npm packages, you need to copy the library over (to where the symlink used to be), and then run a local HTTP server (easiest is to use python for this or a simple http server from npm):
```
cp dist/media-stream-library.min.js examples/browser/
cd examples/browser
python -m SimpleHTTPServer
```

But since you're on Windows, I don't know how easy it is to get hold of a python package. I would recommend you to just clone the repository and start from there:
```
git clone https://github.com/AxisCommunications/media-stream-library-js
cd media-stream-library-js
yarn
yarn examples
```

cubecleveland, 2019-11-22:
super duper all works and i browsed to the camera file and see the html.
now: i enter my camera IP but get the folloowing in the console on page load:
[image]

steabert, 2019-11-25:
Now you ran into the problem that is covered in #176 , so please continue there and try the work-around I posted. If it doesn't work you can continue commenting in that issue.

closed 2019-11-25 by steabert