AxisCommunications/media-stream-library-js #291: Oldest Axis Firmware that supports media-stream-library.js

jjfrankovich, 2019-11-21:
We found an Axis Camera (210A) with Firmware 4.40 does not support media-stream-library.js, we upgraded fully to the newest for that camera which is 4.47.1 and that does not work either. We have some firmware with another camera 6.53.2.1 that works fine. Does anyone know the exact firmware versions where we can start using media-stream-library?
I would guess anything 6.X would be OK?

steabert, 2019-11-22:
I'm pretty sure 6.50 is the minimum for Axis cameras with this library, but I'll have to double check.

steabert, 2020-02-14:
This page: https://www.axis.com/en-us/support/technical-notes/browser-support would indicate that there is WebSocket support on 7.10 and above, but it seems that it already works for you at 6.53. So I would guess that lower than 6.53 wont work.

closed 2020-02-14 by steabert

folkhack, 2020-09-01:
@steabert - We've seen WebSocket support going all the way down to 6.50, but that was dependent on the camera model; ie: we have M3006 cameras that support it running 6.50 firmware. Another way to check for support is looking at the plain config for WebSocket related settings.

Overall I wouldn't assume support for the >=6.50 && <6.53 gap; **BUT** we have successfully tested WebSocket streaming on 6.50 devices. As far as we know this is an unofficially supported feature that likely arrived earlier in the firmware than it's PR announcement.