AxisCommunications/media-stream-library-js #294: Uncaught RangeError: "value" argument is out of bounds

akermen, 2019-11-28:
The browser example seems to give an exception when opened from more than one browser windows.

To reproduce this just clone the master branch and run commands below:
```
git clone https://github.com/AxisCommunications/media-stream-library-js.git .
npm install
yarn build
yarn examples
yarn rtsp
```

Then open the `http://localhost:8080/test/h264.html` address from two browser sessions.

The error message at the console:
```
isom.ts:1345 Uncaught RangeError: "value" argument is out of bounds
    at checkInt (index.js:1259)
    at Uint8Array.writeUInt32BE (index.js:1369)
    at UInt32BE.eval [as copy] (isom.ts:343)
    at Box.copy (isom.ts:1333)
    at Container.copy (isom.ts:1333)
    at Container.copy (isom.ts:1333)
    at Transform.transform [as _transform] (index.ts:147)
    at Transform._read (_stream_transform.js:184)
    at Transform._write (_stream_transform.js:172)
    at doWrite (_stream_writable.js:428)
```

I know the Docker RTSP streaming is meant for testing only but this message may be related to an generic issue not specific to the test stream.

[image]

steabert, 2020-02-06:
I managed to reproduce it now. Will look into it.

ramoncaldeira, 2020-04-22:
@akermen I also faced this exception in my test stream.

It seems that this error happens in the Mp4Muxer component, in the line `moof.copy (data, 0)`.

You can debug the moof structure by calling `moof.print()`. In my case, the `sample_duration` entry inside box trun was negative. It looks like my stream is sending non-monotonic timestamps in its RTP packets.

akermen, 2020-04-23:
Thank you @ramoncaldeira for detailed debug information.

First I thought I could fix by just modifying the test stream generation, but experienced same issue consistently with a live stream from internet (a broadcast tv station) which I have no control of. Unfortunately the live stream appears to be not public anymore.

What I understand is as you point out non-monotonic timestamp although should not happen but at the same time can be expected.

ramoncaldeira, 2020-04-23:
I am no streaming expert, but what about introducing an exponential average in the expression below?

https://github.com/AxisCommunications/media-stream-library-js/blob/d3ec6caec9a057e7cfca499869e14f308c955634/lib/components/mp4muxer/helpers/boxbuilder.ts#L220-L223

I think this way we could smooth out unexpected values.

E.g. 
`beta * (timestamp - trackData.lastTimestamp) + (1-beta) * last_duration`

henryctam, 2020-08-12:
I can reproduce this issue with the latest code(as of today).  @steabert, have you had a chance to look into it? Thanks.

henryctam, 2020-08-31:
@ramoncaldeira @akermen Did you guys ever solve this?  I'm running into the same problem.

akermen, 2020-09-01:
@henryctam, unfortunately no.

steabert, 2021-04-10:
I originally also traced this back to a negative `sample_duration`, but I was unable to reproduce since, and the code has been refactored somewhat since the IDR fix. Is this still reproducible?

akermen, 2021-04-15:
@steabert , Yes, with the main branch as of today as can be seen on the screen capture:
[Screen capture](https://user-images.githubusercontent.com/6176664/114845593-91411d00-9de4-11eb-86b4-76403cabc433.mp4)

commit 61ebe5e0 (2023-09-24): fix: force negative frame durations to zero

Non monotonic streams can produce negative sample duration
and result in RangeError: "value" argument is out of bounds.

This isn't correct behaviour, but by setting the duration
to zero a crash is prevented.

Fixes #294
Fixes #652
PR #769 (2024-04-08): fix: force negative frame durations to zero
Non monotonic streams can produce negative sample duration and result in RangeError: "value" argument is out of bounds.

This isn't correct behaviour, but by setting the duration to zero a crash is prevented.

Fixes #294
Fixes #652
commit e6773e7c (2024-04-08): fix: force negative frame durations to zero (#769)

Non monotonic streams can produce negative sample duration
and result in RangeError: "value" argument is out of bounds.

This isn't correct behaviour, but by setting the duration
to zero a crash is prevented.

Fixes #294
Fixes #652
closed 2024-04-08 by steabert