AxisCommunications/media-stream-library-js #295: Seek

stravi, 2019-11-28:
Hi, I'm trying to make seek working.
I'm using Html5VideoPipeline. I thought it will work out of the box, I will seek using the controls available from HTML5 `<video>` element and it will magicaly make the required calls to the server to adjust the `npt` startTime. Well seems that things are not setup to work like this.
Then I decided to listen on `<video>` element for the play event which is being triggered when a new position in the video is selected.
```
mediaElement.addEventListener('play', (data) => {
//TODO get the desired positionTime
      const positionTime = 402;
      this.pipeline2.rtsp.pause();
      this.pipeline2.rtsp.play(positionTime);
    });
```
But this ain't work as `RtspSession.play()` is sending the request **without** range header ```Range: `npt=${this.startTime || 0}-`,``` as it's being done in the `RtspSession._onSdp()`
Is this a bug or done on purpose? Is using `pipeline.rtsp` object the correct approach to make this happen or do you have a better way for achieving this?

steabert, 2020-02-06:
To answer your first observation, indeed, there is no magical seek functionality. The library is intended to provide the minimum needed to stream video, but doesn't offer any higher level video player functionality so far.
Coupling video element events to the pipeline is exactly how it could be done, and it seems like you did indeed find a bug, it should certainly be possible to send a range via play.

lekoaf, 2020-11-13:
Closing due to inactivity. Feel free to open again if this was an error.

closed 2020-11-13 by lekoaf