AxisCommunications/media-stream-library-js #310: view RTSP stream in browser

532910, 2020-01-14:
I'm looking for a solution to view RTSP stream via browser. For example I can play it with mpv:
`mpv --rtsp-transport=tcp 'rtsp://IP:554/user=admin&password=&channel=1&stream=0.sdp'`

Is `media-stream-library-js` suitable for my case? Should I use it with websockify?

I've tried `examples/browser/test/h264.html` with
```
ws: { uri: `ws://localhost:8854/` },
rtsp: { uri: `rtsp://IP:554/user=admin&password=&channel=1&stream=0.sdp` },
```
in `examples/browser/test/h264-player.js` and `websockify 8854 IP:554` runned on localhost, but it doesn't work for me.

steabert, 2020-01-29:
If you stream RTP over WebSocket, then yes, this library should be suitable for your case. Not that we don't support all of the RTP spec, and that we currently only have H.264/AAC support.

You can start with some details regarding your problem. What is it that doesn't work, do you get an error message?

532910, 2020-01-30:
mpv and ffplay say it is h264 without sound
there are no error in firefox, it just shows nothing, web developer console shows `Empty string passed to getElementById().` but it shows it when websockify is not runned too

```
% websockify 8854 IP:554
WebSocket server settings:
  - Listen on :8854
  - Flash security policy server
  - No SSL/TLS support (no cert file)
  - proxying from :8854 to 10.8.4.78:554
127.0.0.1 - - [30/Jan/2020 04:01:18] 127.0.0.1: Plain non-SSL (ws://) WebSocket connection
127.0.0.1 - - [30/Jan/2020 04:01:18] 127.0.0.1: Version hybi-13, base64: 'False'
127.0.0.1 - - [30/Jan/2020 04:01:18] connecting to: IP:554
127.0.0.1 - - [30/Jan/2020 04:03:22] code 400, message Bad HTTP/0.9 request type ("\x88\x8fUøE9V\x10\x11X'\x9f")
```

I can provide you access to this websocket or even to RTPS socket.

steabert, 2020-02-06:
The odd thing is that nothing in the library or example uses `getElementById`, how exactly do you run the example, with `yarn examples`?

What do you see in the network tab, is there a websocket request?

532910, 2020-02-09:
No yarn.
Steps I've done:
1. put [h264-player.js](https://github.com/AxisCommunications/media-stream-library-js/blob/master/examples/browser/test/h264-player.js), [h264.html](https://github.com/AxisCommunications/media-stream-library-js/blob/master/examples/browser/test/h264.html) and [media-stream-library.min.js](https://github.com/AxisCommunications/media-stream-library-js/releases/download/v6.1.4/media-stream-library.min.js) into local folder (/tmp/rtsp-test)
2. updated `h264-player.js`:
```js
    ws: { uri: `ws://localhost:8854/` },
    rtsp: { uri: `rtsp://IP:554/user=admin&password=&channel=4&stream=0.sdp` },
```
3. started websockify:
`$ websockify 8854 <IP>:554`
4. opened file:///tmp/rtsp-test/h264.html in browser

532910, 2020-02-09:
Yes, I see a websocket request in the network tab:

Request:
```
Host: localhost:8854
User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:72.0) Gecko/20100101 Firefox/72.0
Accept: */*
Accept-Language: en,en-US;q=0.7,ru;q=0.3
Accept-Encoding: gzip, deflate
Sec-WebSocket-Version: 13
Origin: null
Sec-WebSocket-Protocol: binary
Sec-WebSocket-Extensions: permessage-deflate
Sec-WebSocket-Key: gVfw9hwpGtCO6wfGwOojiA==
DNT: 1
Connection: keep-alive, Upgrade
Pragma: no-cache
Cache-Control: no-cache
Upgrade: websocket
```
```
OPTIONS rtsp://IP:554/user=admin&password=&channel=4&stream=0.sdp RTSP/1.0
CSeq: 1
```

Response:
```
HTTP/1.1 101 Switching Protocols
Server: WebSockify Python/3.7.6
Date: Sun, 09 Feb 2020 08:18:34 GMT
Upgrade: websocket
Connection: Upgrade
Sec-WebSocket-Accept: tY+QGsBXZNfQRHOfdx3NMRKgBFM=
Sec-WebSocket-Protocol: binary
```
```
RTSP/1.0 200 OK
Server: H264DVR 1.0
cseq: 1
Public: OPTIONS, DESCRIBE, SETUP, TEARDOWN, GET_PARAMETER, PLAY, PAUSE
```

Updating media-stream-library.min.js to the v7.0.0 I got: `Error: rtsp: expected sequence number media-stream-library.min.js:8:49748`

steabert, 2020-02-14:
I had a quick look at the code (where the error is printed) think it _might_ be that our code is only recognizing `CSeq` and not `cseq`. So it might be that we're not allowing for case insensitive headers, which I think is probably wrong.

532910, 2020-02-14:
I've `%s/CSeq/cseq/g` in `media-stream-library.min.js`, but unfortunately there is a network problem with RTSP host right now. I'll reply if this helps when it become available.

532910, 2020-02-17:
RTSP host is up again and I confirm that replacing `CSeq` with `cseq` in media-stream-library.min.js solves this issue!

532910, 2020-02-28:
v7.0.1 works fine for me

532910, 2020-03-01:
Sorry, my fault, v7.0.1 not fixed!

patricerosay, 2020-03-11:
Hi there,
Thank you @532910 for sharing.
I'm in the same usage and this lib sounds like exactly what I've been looking for for a while.
I understand the lib is designed for Axis IP cameras as they provide RTSP / H264 through WebSockets
I understand this lib can work with a non Axis camera, as long as they provide RTSP/264 stream and the stream is relayed (or controlled ?) on websockets.

@532910 : I understand your IP camera is not Axis and is not providing RTSP stream on WebSocket by nature, and you are using websockfy to emulate this. Is this correct ?

Websocketing is not where i'm the best (understatement...) 
@steabert : I 've search in the documentation for if rtsp streams are passed through WS, why the rtsp address needs to be given in the Html5VideoPipeline ?
This question seem to indicate that I've not understood how it works. 
Could you please explain?

532910, 2020-03-11:
> @532910 : I understand your IP camera is not Axis and is not providing RTSP stream on WebSocket by nature, and you are using websockfy to emulate this. Is this correct ?

https://github.com/tg123/websockify-nginx-module

patricerosay, 2020-03-12:
Thank you for the hint.

the camera is password protected:

rtsp://<user>:<password>@192.168.1.33:554 
<http://admin:Bepeho31@192.168.1.33/cgi-bin/mjpg/video.cgi?channel=0&subtype=1>

This cause a

RTSP/1.0 401 Unauthorized..CSeq:..WWW-Authenticate: Digest realm=...

Do you have an idea of how I could configure websockify to handle this ?

532910, 2020-03-12:
I think it should be on the `media-stream-library-js` side, https://github.com/AxisCommunications/media-stream-library-js/blob/master/examples/browser/test/h264-player.js#L10, not on websockify.

Moreover plaintext authentication has no reason nowadays.

steabert, 2020-03-27:
> Websocketing is not where i'm the best (understatement...)
> @steabert : I 've search in the documentation for if rtsp streams are passed through WS, why the rtsp address needs to be given in the [...]

The WebSocket connection is just used as a tunnel to the RTSP server, so we still need the RTSP URI to construct the message that is sent to the actual RTSP server. However, the hostname in the RTSP URI doesn't matter any more at that point.

closed 2020-04-14 by steabert

532910, 2020-05-08:
Just updated to 7.0.4 and confirm it works fine, thank you Steven!

henryctam, 2020-08-15:
I was also trying to find out how one could supply the username and password to a protected rtsp stream.  After going through some of the code, I found the answer.  Just add this parameter to the player:

auth: { username: "user username", password: "your password },

commit 78f589e3 (2020-04-14): fix: allow lower-case header names as per RFC spec

Fixes: #310