AxisCommunications/media-stream-library-js #313: How to get client stream information

buntin, 2020-01-24:
How can I get the "Client stream information" that's displayed on the live view for Axis encoders?  Is that a feature of this library?  Or somewhere in the MSE?  I can't find it anywhere.

I'm particularly interested in the "Bitrate" and "Buffer" values.

rikteg, 2020-01-27:
A quick comment. The different statistics shown in "Client stream information" is
derived from a couple of different places.

We use the progress-event to know when to update the statistics.
https://developer.mozilla.org/en-US/docs/Web/API/HTMLMediaElement/progress_event

Bitrate and framerate comes from the pipeline, that is: `pipeline.bitrate` and
`pieline.framerate` which both are arrays of the respective value of each track (e.g. video and audio).

Buffer comes from the video element itself. See: https://developer.mozilla.org/en-US/docs/Web/API/HTMLMediaElement/buffered
E.g.: `document.querySelector('video').onprogress = x => console.log(x.srcElement.buffered.start(0), x.srcElement.buffered.end(0))`

And lastly, "Dropped Frames" comes from getVideoPlaybackQuality. See:
https://developer.mozilla.org/en-US/docs/Web/API/HTMLVideoElement/getVideoPlaybackQuality

buntin, 2020-01-28:
Thanks so much for the links and example code.  I was able to acquire the stream statistics I was looking for.

closed 2020-01-28 by buntin