AxisCommunications/media-stream-library-js #329: Websocket absolutely required? Safari supported?

MarcelPawlo, 2020-02-13:
Is  my assumption correct that the device from which the RTSP stream is to  be retrieved necessarily has a web socket available? We have to use Axis  cameras as well as our bodycam, for example, which does not meet this  requirement.

I also saw that neither Safari browser nor bidirectional audio is available, is that correct?

steabert, 2020-02-14:
Hi! The device itself (RTSP server) does not necessarily need to have WebSocket support, but then you need a WebSocket proxy in-between. So if you directly want to access the device with a browser, the device itself will need a WebSocket server.

The library should work on any browser that supports WebSocket and MediaSource extensions, although codec support will depend on the browser. Safari should work though (note that iOS Safari is a different beast and did not have MediaSource support until recently).

The audio/video components in the current library currently only work in the direction from server to browser. There is no audio/video recording in the browser on the client side.

532910, 2020-02-28:
Moreover: all iOS browsers (firefox, chrome) ARE Safari. So it will not work on iOS at all.

532910, 2020-02-28:
> did not have MediaSource support until recently

https://developer.mozilla.org/en-US/docs/Web/API/MediaSource/MediaSource
Steven, what is **recently**?

steabert, 2020-03-27:
https://caniuse.com/#feat=mediasource

closed 2020-03-27 by steabert

532910, 2020-03-29:
https://downforeveryoneorjustme.com/caniuse.com

> It's not just you! caniuse.com is down.