AxisCommunications/media-stream-library-js #340: sometimes playback stops with Error: Unknown chunk of length

532910, 2020-02-28:
Overall it works fine, but sometimes playback stops accidentally or just doesn't start (stops on start)
 
Firefox console shows:
```
Error: Unknown chunk of length 1225 media-stream-library.min.js:8:163325
    value https://domain.tld/media-stream-library.min.js:8
    value https://domain.tld/media-stream-library.min.js:8
    transform https://domain.tld/media-stream-library.min.js:8
    _read https://domain.tld/media-stream-library.min.js:8
    _write https://domain.tld/media-stream-library.min.js:8
    g https://domain.tld/media-stream-library.min.js:8
    a https://domain.tld/media-stream-library.min.js:8
    write https://domain.tld/media-stream-library.min.js:8
    y https://domain.tld/media-stream-library.min.js:8
    apply self-hosted:1910
    emit https://domain.tld/media-stream-library.min.js:8
    _ https://domain.tld/media-stream-library.min.js:8
    w https://domain.tld/media-stream-library.min.js:8
    push https://domain.tld/media-stream-library.min.js:8
    onmessage https://domain.tld/media-stream-library.min.js:8
```

**To Reproduce**
I just open `examples/browser/test/h264.html` (with `ws` and `rtsp` modified in `h264-player.js`) and see a picture from the cam, but it's just not updated.

**Environment**
 - Browser fresh Firefox or Chrome
 - Version v7.0.1
Two setups are tested with the same issue:
1. Debian Sid with local files without webserver + websockify 0.8.0 on the same host
2. Debian Sid as frontend (browser) + Debian Buster with nginx + websockify-nginx-module master

532910, 2020-02-28:
mpv always works fine and has no such issue

steabert, 2021-04-11:
We don't support the entire RTP spec, so that can definitely be a difference with mpv.

Closing for now, as old and cannot reproduce myself. However, if you can find out more info on exactly what the unknown chunk is, please reopen (I can help with setting up a debugging environment if necessary).

closed 2021-04-11 by steabert