AxisCommunications/media-stream-library-js #390: Trying to run examples/browser/test/h264.html

deam79, 2020-04-24:
Hi.

I'm trying to run the examples, and specifically those from the browser/test directory.

I have the server running with "yarn examples", and am also running "yarn rtsp" in a separate terminal window. I can view the RTSP stream produced with "vlc rtsp://0.0.0.0:8554/test", but when I browse to the h264.html page, I'm not getting video.

The browser console outputs "Firefox can't establish a connection to the server at ws://127.0.0.1:8854/"

There's a tcp-ws-proxy.log file produced, at the top of which I've got "Error: Cannot find module '../dist/cjs/index.node.js'"

Any ideas? Rather feel like I'm probably missing something obvious here...

mmeyers-solartech, 2020-04-24:
Did "Yarn examples" work for you right out the gate? I installed with yarn add media-stream-library but running yarn examples says webpack isn't configured, and when I add it I get some major compile problems.

deam79, 2020-04-24:
I did go through a couple of steps as I installed yarn. The following seemed to work for me:

First, enabling and adding the yarn repository:

curl -sS https://dl.yarnpkg.com/debian/pubkey.gpg | sudo apt-key add -
echo "deb https://dl.yarnpkg.com/debian/ stable main" | sudo tee /etc/apt/sources.list.d/yarn.list

Then just the more normal:

sudo apt update
sudo apt install yarn
yarn add media-stream-library

steabert, 2020-04-24:
Hi @deam79 , the current instructions are actually meant for those cloning the repository and working with it for development. I've already commented in another issues that the current README is not helpful.

If you clone the repo, run `yarn`, then `yarn build`, and then `yarn examples` parallel with `yarn rtsp`, you should be able to get going. There are more people with a similar problem, so I blame our documentation. I'll try to go through it ASAP it and fix any problems with both the text and any bugs in the examples.

mmeyers-solartech, 2020-04-24:
Ahh yarn build.....Thanks again Steabert!

deam79, 2020-04-24:
Ah, great. Thanks for that, I'll go and give that a whizz!

deam79, 2020-04-24:
Excellent, success! Thanks for that.

closed 2020-10-10 by steabert