AxisCommunications/media-stream-library-js #391: Unable to run examples

mmeyers-solartech, 2020-04-24:
Hi there, I'm trying to follow the instructions to host the examples on my machine. I am trying to follow the instructions exactly. 
1. yarn add media-stream-library. This works, it adds the project.
2. yarn examples (from top folder). Examples not found. Okay no big deal, so I navigate to example folder
3. yarn examples (from /node_modules/media-stream-library). Error, webpack: not found. 
4. yarn add webpack --dev (from same folder as above). Seems to work. 
5. yarn examples. I get a ton of compile errors. The first few will be pasted below. 

What am I doing wrong here? I have an RTSP feed ready to play and I want to test it but can't get the examples up and running. Any help is appreciated, thank you.

ERROR in ./lib/utils/protocols/sdp.ts 4:0
Module parse failed: The keyword 'interface' is reserved (4:0)
You may need an appropriate loader to handle this file type, currently no loaders are configured to process this file. See https://webpack.js.org/concepts#loaders
| import { NtpSeconds, seconds } from './ntp'
| 
> interface ConnectionField {
|   // c=<nettype> <addrtype> <connection-address>
|   networkType: 'IN'
 @ ./lib/utils/protocols/index.ts 1:0-21 1:0-21
 @ ./lib/utils/index.browser.ts
 @ ./lib/index.browser.ts

ERROR in ./lib/components/canvas/index.ts 8:0
Module parse failed: The keyword 'interface' is reserved (8:0)
You may need an appropriate loader to handle this file type, currently no loaders are configured to process this file. See https://webpack.js.org/concepts#loaders
| import { VideoMedia } from '../../utils/protocols/sdp'
| 
> interface BlobMessage {
|   readonly blob: Blob
|   readonly ntpTimestamp: number
 @ ./lib/components/index.browser.ts 5:0-24 5:0-24
 @ ./lib/index.browser.ts

ERROR in ./lib/components/h264depay/index.ts 10:23
Module parse failed: Unexpected token (10:23)
You may need an appropriate loader to handle this file type, currently no loaders are configured to process this file. See https://webpack.js.org/concepts#loaders
| export class H264Depay extends Tube {
|   constructor() {
>     let h264PayloadType: number
|     let idrFound = false
| 
 @ ./lib/components/index.browser.ts 6:0-27 6:0-27
 @ ./lib/index.browser.ts

ERROR in ./lib/components/jpegdepay/index.ts 10:23
Module parse failed: Unexpected token (10:23)
You may need an appropriate loader to handle this file type, currently no loaders are configured to process this file. See https://webpack.js.org/concepts#loaders
| export class JPEGDepay extends Tube {
|   constructor() {
>     let jpegPayloadType: number
|     let packets: Buffer[] = []
|     let jpegDepay: (
 @ ./lib/components/index.browser.ts 8:0-27 8:0-27
 @ ./lib/index.browser.ts

ERROR in ./lib/pipelines/rtsp-mjpeg-pipeline.ts 11:24
Module parse failed: Unexpected token (11:24)
You may need an appropriate loader to handle this file type, currently no loaders are configured to process this file. See https://webpack.js.org/concepts#loaders
|  */
| export class RtspMjpegPipeline extends RtspPipeline {
>   constructor(rtspConfig?: RtspConfig) {
|     super(rtspConfig)
| 
 @ ./lib/pipelines/index.browser.ts 5:0-37 5:0-37
 @ ./lib/index.browser.ts

ERROR in ./lib/components/basicdepay/index.ts 12:28
Module parse failed: Unexpected token (12:28)
You may need an appropriate loader to handle this file type, currently no loaders are configured to process this file. See https://webpack.js.org/concepts#loaders
| 
| export class BasicDepay extends Tube {
>   constructor(rtpPayloadType: number) {
|     if (rtpPayloadType === undefined) {
|       throw new Error('you must supply a payload type to BasicDepayComponent')
 @ ./lib/components/index.browser.ts 4:0-28 4:0-28
 @ ./lib/index.browser.ts

steabert, 2020-04-24:
Hi!

You cannot run the examples when you install it as a node module. We refer to the examples so that you get an idea of how it could be used, but I see now that the README is completely confusing! I've opened an issue to fix the README.md file and provide some clarification how to run examples.

Thanks!

mmeyers-solartech, 2020-04-24:
Hi, thanks for an incredibly quick reply. Looks like you know the problem. I'll keep my eye out for a README update. I appreciate this. Is there an alternative command to install that allows for running examples? (Ignore if you're throwing this in the README at some point)

steabert, 2020-04-24:
Are you using Windows or Linux ?

If you use Linux, you can get the examples running by cloning this repo, then run `yarn`, `yarn build`, and then `yarn examples`. Then you can run examples against an IP camera. For test streams with RTSP, one can run `yarn rtsp` separately to get up an RTSP server, but then you need to have a working docker setup.

The current development environment is focused on Linux, but if you use Windows let us know if you run into any problems.

steabert, 2020-04-24:
And as I said, I'll go through the whole procedure as if I would use it externally, and fix anything on the way. You can expect an update within a few days, and then we can see if you can get up and running.

mmeyers-solartech, 2020-04-24:
I'm using VLC rather than a functional docker setup to stream locally at 127.0.0.1, it should hypothetically be the same functionality with VLC functioning as the media server, I'm just unable to view the streams

steabert, 2020-05-04:
Still working on this. Have you made any progress so far?

PR #543 (2021-01-13): feat: copy minified file to example
Instead of using a symlink, which doesn't work on Windows, we copy the minified file to the examples folder.

Fixes #412
Fixes #391
Fixes #176
commit 2ff3a611 (2021-01-13): feat: copy minified file to example

Instead of using a symlink, which doesn't work on
Windows, we copy the minified file to the examples
folder.

Fixes #412
Fixes #391
Fixes #176
closed 2021-01-13 by steabert