AxisCommunications/media-stream-library-js #392: Playing RTSP stream produced by VLC

mmeyers-solartech, 2020-04-24:
I'm running the examples now using yarn examples. I currently have VLC streaming an RTSP live feed at "rtsp://127.0.0.1:8554/stream". I can view it in another VLC window or my swing app or wherever, so I know the output is solid. It's H.264 no transcoding (originating from an axis camera). 

All my attempts to plug that localhost URL into the "simple.html" page refuse. I also have a react app where I'm attempting the same thing. The output in the firefox console is this:

Firefox can’t establish a connection to the server at ws://127.0.0.1:8554/stream/rtsp-over-websocket. media-stream-library.min.js line 518 > eval:60:15
failed to load token 404 <empty string> openwebsocket.ts:76:21
uncaught exception: Object

It does seem strange that "rtsp-over-websocket" is being appended to the URL, I can see where it's happening in openwebsocket.ts, but that's probably unrelated. Any ideas why it can't access that localhost stream?

steabert, 2020-04-27:
Do you have a WebSocket proxy running that forwards the RTSP stream? That's what `yarn rtsp` does, it both starts an RTSP server, but also starts a WebSocket proxy to it.

The `simple-player.js` that is used for the `simple.html` example sets both the WebSocket and RTSP URIs, and they default to those used by an Axis camera. You'll have to modify that file to use the endpoints that you are using.

jianjunchang, 2020-06-17:
I have encountered the same problem, and i found the vlc is not support transport: rtp/avp/tcp, so response a 461 error status when sending a 'setup' request from rtsp client, so i use live555mediaserver as the rtsp server instead, but the live555mediaserver only support ".ts",".264",".m4e", the "mp4" is not support what is i using.

mmeyers-solartech, 2020-06-18:
Unrelated but I ended up using node-rtsp-stream and jsmpeg player for a web viewer. It's powerful and displays my video nicely.

steabert, 2020-08-02:
Great that you found a solution that works for you!

closed 2020-08-02 by steabert