AxisCommunications/media-stream-library-js #395: The dependency towards yargs

mattias-kindborg-axis-communications, 2020-04-30:
I notice the package has a dependency towards `yargs`, but `yargs` is only used in `examples` and `rtsp-ws-server`. I propose removing this dependency from the main package, and add two new packages, one in `examples` and one in `rtsp-ws-server`, each with this dependency.

If you agree, I would happily contribute a PR.

steabert, 2020-04-30:
We've had some thoughts about using a monorepo, and I was planning to have a look again at our README.md which isn't all that helpful. Are you proposing a monorepo, or did you have another solution in mind? If the PR isn't too much work, you can just put it up and we can discuss it there.

mattias-kindborg-axis-communications, 2020-04-30:
No, I don't propose a monorepo. A monorepo, as I know it, is a repository with multiple packages that all are relevant to the client and his or her production code. [Angular](https://github.com/angular/angular) is a good example, where each package in the repository is published as a separate npm package.

I only propose to create new packages for the samples in this repository, and correct me if I'm wrong in thinking that `rtsp-ws-server` also is something of a sample, and not being a part of what the client consumes after installing this NPM package. Removing `yargs`, or at least moving it to be a dev dependency, would better describe the required dependencies of the package.

steabert, 2020-04-30:
Yes, dev dependency would be appropriate, I just didn't understand the "add two new packages" part.

PR #397 (2020-04-30): Move yargs to dev dependencies
Closes #395
closed 2020-04-30 by steabert