AxisCommunications/media-stream-library-js #396: Missing dependency towards md5.js

mattias-kindborg-axis-communications, 2020-04-30:
**Describe the bug**
When running this in Node.js using JavaScript and CommonJS modules, I get the following runtime error.

```
internal/modules/cjs/loader.js:983
  throw err;
  ^

Error: Cannot find module 'md5.js'
Require stack:
- .../node_modules/media-stream-library/dist/cjs/components/auth/digest.js
```

I think the `md5.js` `devDependencies` should be moved to `dependencies`. This will solve the issue for Node.js, but I don't know whether it will cause another issue for other runtimes.

**Environment (please complete the following information):**
 - OS: iOS
 - Browser Node.js v12.16.2
 - Version 7.0.3

closed 2020-05-04 by steabert

reijosirila, 2020-12-03:
This module dependency creates issues using esm / webpack in some conditions. I'll try to debug it more but here's a trace

`
Uncaught TypeError: Cannot set property 'needReadable' of undefined
    at MD5.Transform$1 (module.js?13e5:12947)
    at MD5.HashBase (module.js?13e5:13100)
    at new MD5 (module.js?13e5:13280)
    at new DigestAuth (module.js?13e5:13496)
    at Transform.eval [as _transform] (module.js?13e5:13586)
    at Transform._read (module.js?13e5:5298)
    at Transform._write (module.js?13e5:5280)
    at doWrite (module.js?13e5:4928)
    at writeOrBuffer (module.js?13e5:4928)
    at Transform.Writable.write (module.js?13e5:4850)
Transform$1 @ module.js?13e5:12947
HashBase @ module.js?13e5:13100
MD5 @ module.js?13e5:13280
DigestAuth @ module.js?13e5:13496
eval @ module.js?13e5:13586
Transform._read @ module.js?13e5:5298
Transform._write @ module.js?13e5:5280
doWrite @ module.js?13e5:4928
writeOrBuffer @ module.js?13e5:4928
Writable.write @ module.js?13e5:4850
ondata @ module.js?13e5:4285
emitOne @ module.js?13e5:172
emit @ module.js?13e5:266
readableAddChunk @ module.js?13e5:3961
Readable.push @ module.js?13e5:3910
Transform.push @ module.js?13e5:5250
eval @ module.js?13e5:6201
transform @ module.js?13e5:6201
Transform._read @ module.js?13e5:5298
Transform._write @ module.js?13e5:5280
doWrite @ module.js?13e5:4928
writeOrBuffer @ module.js?13e5:4928
Writable.write @ module.js?13e5:4850
ondata @ module.js?13e5:4285
emitOne @ module.js?13e5:172
emit @ module.js?13e5:266
readableAddChunk @ module.js?13e5:3961
Readable.push @ module.js?13e5:3910
WSSource.socket.onmessage @ module.js?13e5:9990
`

reijosirila, 2020-12-04:
I don't know, is it good idea to use all this nodejs module dependency tree: 

https://github.com/nodejs/readable-stream/blob/1448d473d9964ea87612070fe94d2dc6565e50f3/lib/_stream_transform.js#L110 

library on a browser build? 

Maybe it's good idea to use some drop-in replacement like this one: https://www.npmjs.com/package/ts-md5 ?

PR #402 (2020-05-04): Move md5.js to dependencies
The dependency `md5.js` is noted in `devDependencies`, but it is actually a production dependency since it is referenced from the class `DigestAuth`.

Closes #396
commit bdb41a36 (2020-05-04): fix(deps): move md5.js to dependencies

The dependency md5.js is noted in devDependencies, but it is actually a
production dependency since it is referenced from the class DigestAuth.

Closes #396