AxisCommunications/media-stream-library-js #412: Trying to run examples/browser/camera/simple.html

mas1515, 2020-05-26:
Hello @steabert

i want test this file and my step was 

1- download https://github.com/AxisCommunications/media-stream-library-js
2- cd media-stream-library-js
3- run npm install     to install all package
4- yarn examples       it works and show this message  
yarn run v1.22.0
$ http-server examples/browser
Starting up http-server, serving examples/browser
Available on:
  http://192.168.1.118:8080
  http://192.168.52.1:8080
  http://192.168.199.1:8080
  http://192.168.1.106:8080
  http://127.0.0.1:8080

5- after that i go to this address 
http://localhost:8080/camera/simple.html

and enter my camera ip address
click play button 

but nothing happened  i check console in chrome
and say  
1-  Uncaught SyntaxError: Unexpected token '.    media-stream-library.min.js:1' 

2- Uncaught TypeError: Cannot destructure property 'pipelines' of 'window.mediaStreamLibrary' as it is undefined.        http://simple-player.js:1/

what is problem ????

!!! i can see my  ip  camera  in vlc  with this address and it is ok  
rtsp://admin:test@192.168.1.64:554/cam/realmonitor?channel=1&subtype=0
.

xieyubo, 2020-09-28:
Copy "media-stream-library.min.js" from "dist/" into "examples/browser" should fix this issue.

lekoaf, 2020-10-12:
Can confirm this as well. It seems like the symlinks between the examples folder and the minified file doesn't work very well (at all) on Windows. It works on Linux though.

PR #543 (2021-01-13): feat: copy minified file to example
Instead of using a symlink, which doesn't work on Windows, we copy the minified file to the examples folder.

Fixes #412
Fixes #391
Fixes #176
commit 2ff3a611 (2021-01-13): feat: copy minified file to example

Instead of using a symlink, which doesn't work on
Windows, we copy the minified file to the examples
folder.

Fixes #412
Fixes #391
Fixes #176
closed 2021-01-13 by steabert