AxisCommunications/media-stream-library-js #418: Chrome pauses stream when tab is changed

MarcelPawlo, 2020-06-24:
Using "examples/browser/camera/simple.html" and playing a RTSP stream, it will stop when I change the tab in Chrome 83. When switching back, you can see that the timestamp hasn't continued.

Any suggestions?

steabert, 2020-08-02:
Yes, you need to handle pause/resume in the player, this library just provides the lower level functionality. You can look at https://github.com/AxisCommunications/media-stream-player-js/blob/master/lib/Player.tsxfor an example how to do this (by checking the `visibilityState`).

closed 2020-08-02 by steabert

MarcelPawlo, 2020-08-04:
Thanks for the tip with the advice with the visibilityState.
Don't get me wrong, but I was looking for a way to NOT pause the player when changing the tab with Chrome. My solution is to establish a new RTSP pipeline like this: `document.addEventListener("visibilitychange", () => if (!(document.visibilityState === "hidden" && document.hidden)) createRtspPipeline(videoNode), false);`

piyush-vvdn, 2021-06-23:
I am also facing the same problem, can you tell me, how to use this?