AxisCommunications/media-stream-library-js #439: Video freezes in the first frame when running in Google Chrome

henryctam, 2020-08-12:
**Describe the bug**
When I run the example http://localhost:8080/test/h264.html in Google Chrome(tested on the latest Chrome version 84.0.4147.105), it only renders the first video frame and then freezes(so the clock on the screen stops at 0:00:00.000).  This problem doesn't appear when running the same example in FireFox or Microsoft Chrome.

**To Reproduce**

Steps to reproduce the behavior:
1. Clone the repo.
2. yarn
3. yarn build
4. yarn examples (to start the web server)
5. yarn rtsp  (to start the rtsp server in docker)
6. Note that examples\browser\media-stream-library.min.js only has a 1-line content that points to ..\..\dist\media-stream-library.min.js and the browser would complain if you load http://localhost:8080/test/h264.html.  So I had to replace the content of examples\browser\media-stream-library.min.js with dist\media-stream-library.min.js. (btw, **is this a bug?**)
7. Now open Chrome to browse to http://localhost:8080/test/h264.html.
8. The browser loads the page and renders the first frame of the video.  The video doesn't progress.

**Screenshots**
[image]

**Environment (please complete the following information):**
 - OS: Windows 10
 - Browser: Google Chrome version 84.0.4147.105

**Additional context**
Add any other context about the problem here.

funman203, 2020-10-03:
Maybe linked to Chromes Autoplay policy cf https://developers.google.com/web/updates/2017/09/autoplay-policy-changes

Chrome does not autoplay unless the video tag has muted attribute or the user has interacted with DOM (onclick for instance)

steabert, 2020-10-10:
As @funman203  explained, this can be an autoplay issue.

The examples in this library are rather low-level, and there isn't much documentation about browser-specific quirks relating to playback itself (since this library is mostly about delivering video data). To get better examples on how to control playback, you can look at the [media-stream-player](https://github.com/AxisCommunications/media-stream-player-js) repository. The latter includes a web component that uses muted autoplay, which should work in all browsers.

closed 2020-10-10 by steabert