AxisCommunications/media-stream-library-js #488: stream transform is not a constructor error

jamesfarquhar1964, 2020-09-21:
When I try to use the Html5CanvasPipeline in my Angular 10 application I get the error below in my browser console:

core.js:4197 ERROR TypeError: stream__WEBPACK_IMPORTED_MODULE_1__.Transform is not a constructor
    at new RtspParser (index.js:21)
    at new RtspPipeline (rtsp-pipeline.js:19)
    at new RtspMjpegPipeline (rtsp-mjpeg-pipeline.js:10)
    at new Html5CanvasPipeline (html5-canvas-pipeline.js:28)
    at CameraControlComponent.PlayVideo (CameraControl.component.ts:37)
    at CameraControlComponent_Template_button_click_5_listener (CameraControl.component.html:11)
    at executeListenerWithErrorHandling (core.js:14310)
    at wrapListenerIn_markDirtyAndPreventDefault (core.js:14345)
    at HTMLButtonElement.<anonymous> (platform-browser.js:582)
    at ZoneDelegate.invokeTask (zone-evergreen.js:399)
defaultErrorLogger @ core.js:4197

I don't know if this is a bug, or an error on my part. "ng serve" did successfully compile my application, so it only seems to be chrome that is unhappy. if it helps here is my typescript that is creating the Html5CanvasPipeline:

import { Component, OnInit } from '@angular/core';
import { Router, ActivatedRoute } from '@angular/router';
import { IPCamera } from '../../Models/camera';
import { CameraControlService } from '../../Services/CameraControl';
import { Html5CanvasPipeline } from 'media-stream-library/dist/esm/pipelines/html5-canvas-pipeline';
import { Observable } from 'rxjs';
import { Transform } from 'stream';

@Component({
  selector: 'app-camera-control',
  templateUrl: './CameraControl.component.html',
  styleUrls: ['./CameraControl.component.scss']
})
export class CameraControlComponent {

  public cameras: IPCamera[]; 
  public cameraList: IPCamera[];
  public currentCamera: IPCamera;
  public pipeline: Html5CanvasPipeline;
  
  constructor(private cameraService: CameraControlService) {
    this.cameraService.ListCameras().subscribe(
      data => { this.cameras = data;},
      err => { console.error('Unable to get list of cameras');}
    )
    
  }

  public PlayVideo() {
    
    if (this.pipeline !== undefined)
    {
      this.pipeline.close();
    }

    let mediaElement = <HTMLCanvasElement> document.getElementById('CameraCanvas');
    this.pipeline = new Html5CanvasPipeline({
      ws: { uri: `ws://' + this.currentCamera.CameraIPAddress + '/rtsp-over-websocket` },
      rtsp: { uri: `rtsp://' + this.currentCamera.CameraIPAddress + '/axis-media/media.amp?videocodec=jpeg` },
      mediaElement,
    })
    this.pipeline.ready.then(() => {
      this.pipeline.play()
    })
  }
}

steabert, 2020-10-10:
We use the `stream` and `buffer` packages from NodeJS, e.g. the `Transform` class is imported from `stream`. Because these do not exist in the browser, the bundler needs to automatically replace them with `browserify` alternatives. Webpack does this automatically when building for target `web`. I'm completely ignorant about Angular, so I'm not sure how to get Angular to automatically replace these imports and use `browserify`.

As an alternative, you could try to import from our pre-built bundle, `dist/media-stream-library.min.js`.

lekoaf, 2020-11-13:
@jamesfarquhar1964, you could perhaps also try the web component in the [media-stream-player](https://github.com/AxisCommunications/media-stream-player-js) library. I'm not sure how well it plays with an Angular application though. Haven't written any Angular code since v1.5ish. :see_no_evil:

steabert, 2021-02-02:
I've changed the library so that we export a bundle by default since v9. So you could try again with the latest version, in which case Angular should just be able to use the pre-bundled stuff.

closed 2021-04-11 by steabert