AxisCommunications/media-stream-library-js #512: support H264 in the future will be exciting

wst-casd, 2020-10-14:
This tool is so fascinating. But when the camera video codec is h264, this tool can't handle that. If it can support h264, it will more useful!

lekoaf, 2020-10-21:
Yeah, that would be cool. Except that the browser support for h.265 is severely lacking.

https://caniuse.com/?search=h265

closed 2020-11-13 by lekoaf

ggregory8, 2022-06-16:
Just revisiting this topic now h265 is supported on a few browsers i.e Edge. Does the media-stream-library handle H265 RTSP streams yet? I can't see any reference to this in the docs yet.

lekoaf, 2022-06-16:
> Just revisiting this topic now h265 is supported on a few browsers i.e Edge. Does the media-stream-library handle H265 RTSP streams yet? I can't see any reference to this in the docs yet.

Nope, we haven't looked at H.265 support yet.