AxisCommunications/media-stream-library-js #524: Help getting a feed

cubecleveland, 2020-10-26:
HI Guys,

I posted here about a year a go asing request abou the llibrary and componenets. We ended up using a third party vendor, but now this came into play again.
We are trying to integrate feeds from working Axis cameras.
The IP and RSTP servers are forwarded and can be accessed via the open internet.
I can get a feed by simply going to a url: http://[my-url]:[my-port]/axis-cgi/mjpg/video.cgi

Again, sorry for the novice question but im having issues understaing how to get anytihng to work here.
1) does this example can be run over the browser... or doest it need a Node server in the background...
2) if it does run via the browser.... is there an easy way to simply 'get a feed' from a url ?
Sorry for what might seems a 'silly' question but the docs are not self explanator...
any input on how we can move forward with this.

Thanks!

lekoaf, 2020-10-26:
Maybe the Player is something you can use instead?

It's a React implementation of this library.

https://github.com/AxisCommunications/media-stream-player-js

lekoaf, 2020-10-27:
Take a look at https://github.com/AxisCommunications/media-stream-player-js/pull/68.
I added an example of how you can show multiple streams in one page.

closed 2020-11-13 by lekoaf