AxisCommunications/media-stream-library-js #534: How to integrate media-stream-library inside Angular8 app

JackZhai2018, 2020-11-05:
When I try to use Html5VideoPipeline in my Angular 8 application , I get the error as following.
Anybody had ever integrate media-stream-library inside angular application successfully?
Can you give a sample for me.

----------------
import {components, pipelines} from 'media-stream-library';
const pipeline = new pipelines.Html5VideoPipeline({

----------------
ERROR TypeError: Cannot read property 'Html5VideoPipeline' of undefined

lekoaf, 2020-11-13:
I think you need to go deeper. `import {pipelines} from 'media-stream-library/dist/esm/index.browser'`.

steabert, 2021-02-02:
It depends how Angular bundles stuff, unfortunately I don't know anything about Angular. If you import like @lekoaf wrote, you'll need to bundle the browserify parts yourself.

closed 2023-03-18 by steabert