AxisCommunications/media-stream-library-js #546: Stream rtsp over websocket and save into file with Node CLI

JackZhai2018, 2020-11-18:
I try to stream rtsp over websocket and save into file with Node CLI.
so I create a new pipeline, can it work for me?

----------------------------------------------------
class CliWsMp4Pipeline extends rtsp_mp4_pipeline_1.RtspMp4Pipeline {
    constructor(config) {
        const { ws: wsConfig, rtsp: rtspConfig, auth: authConfig, file: fileConfig} = config;
        super(rtspConfig);
        if (authConfig) {
            const auth = new auth_1.Auth(authConfig);
            this.insertBefore(this.rtsp, auth);
        }
		
	var accessfile = fs.createWriteStream(file.path);
        const dataSaver = process.stdout.isTTY
            ? (msg) => console.log(msg.type, msg.data)
            : (msg) => process.stdout.pipe(accessfile).write(msg.data);
        const videoSink = component_1.Sink.fromHandler(dataSaver);
		
        const waitForWs = ws_source_1.WSSource.open(wsConfig);
        this.ready = waitForWs.then((wsSource) => {
            wsSource.onServerClose = () => {
                this.onServerClose && this.onServerClose();
            };
            this.prepend(wsSource);
	    this.append(videoSink);
            this._src = wsSource;
        });
    }
}
----------------------------------------------------------------------------------------------------

JackZhai2018, 2020-12-21:
Anyone know how to write the init segment into mp4 file with commponents provided in this library?

steabert, 2021-02-02:
If you want to write MP4 to a file, you'll need to add the RTSP session and parser, H.264 and AAC depay, and MP4 muxer components. The same are used for the html5 video pipeline.

closed 2023-03-18 by steabert